sipeed/NanoKVM #399: Sketchy DNS Servers

personjs, 2025-03-20:
Somewhere in your binaries or init.d options you are overriding DNS nameservers received from DHCP and statically adding 192.168.0.1, 8.8.8.8, 8.8.4.4, 114.114.114.114, and 119.29.29.29. I tried setting it in both /etc/resolv.conf, /etc/init.d/S30eth, and /kvmapp/system/init.d/S30eth but the service still overwrote what my DHCP server mapped them to.

Jonher937, 2025-03-20:
https://github.com/sipeed/NanoKVM/issues/311#issuecomment-2658954062

dormancygrace, 2026-09-14:
Hi! I've included a fix for this in [NanoKVM OS](https://github.com/dormancygrace/NanoKVM-OS), my independent community firmware for SG2002-based NanoKVM Cube and PCIe.

DHCP DNS servers are published through openresolv as plain nameserver entries, with separate lease records per interface. The Web UI also lets you choose DHCP DNS or explicit manual DNS servers, instead of having to edit a resolver file that is overwritten at boot.

A deliberately configured /boot/resolv.conf remains a static override. DNS lease publication, removal and that override passed isolated regression checks.

[Implementation / details](https://github.com/dormancygrace/NanoKVM-OS/blob/9f65f9554142db4a61ba866b4498363f4f21d711/firmware/buildroot/board/overlay/usr/share/udhcpc/default.script.d/50-resolvconf).

If you're still experiencing this issue, you're welcome to try it and let me know whether it resolves the problem on your setup.

It's currently in beta. Development testing has been on PCIe/UXC; Cube testing is still pending. Installing from stock firmware requires flashing a full SD image, so please back up your card first.

[Latest release](https://github.com/dormancygrace/NanoKVM-OS/releases/latest) · [Installation guide](https://github.com/dormancygrace/NanoKVM-OS/blob/main/docs/INSTALL.md)

Please include your board revision and reproduction steps when reporting results. Thanks!