sipeed/NanoKVM #401: Broken interface after update to nanokvm@2.2.4

abuss, 2025-03-21:
After update form 2.2.2 to 2.2.4, the login screen shows up, but after login the following is displayed

[image]

And all the controls are not available

fenio, 2025-03-21:
Seems that 2.2.4 is completely broken. I'm loosing connectivity to NanoKVM over and over again. How to downgrade to 2.2.2?

superphreak4, 2025-03-22:
To downgrade
https://wiki.sipeed.com/hardware/en/kvm/NanoKVM/faq.html#If-there-are-network-disconnections-or-other-abnormal-situations-during-the-update%2C-it-may-lead-to-a-failed-update.-Please-refer-to-the-following-solutions%3A

The command for downgrade I believe is missing a space between directories "/old/kvmapp"
So it would be 
`rm -r /kvmapp && cp -r /root/old/ / && mv /old /kvmapp && reboot`

xST4Rx, 2025-03-23:
rm -r /kvmapp && cp -r /root/old/ / && mv /old /kvmapp && reboot is working 

2.2.4 is broken ...