sipeed/NanoKVM #403: After update 2.2.4 Windows 11 after sleep, USB failed with "A request for the USB device descriptor failed."

mckenm, 2025-03-22:
After I updated my NanoKVM Full to 2.2.4, it start fail to recognize by Windows 11 after it went to sleep.  This make this IP-KVM unusable, since the keyboard and mouse is no longer accepted as input.

I tried use external USB (AUX) power and just use the USB (HID) both method result with same issue.

When AUX is use and I plug/unplg the HID USB, it keep having issue with W11, I have to power cycle it to have it works again.

[image]

mckenm, 2025-03-22:
Is there any way I can restart the USB through terminal to make it work again?

superphreak4, 2025-03-22:
USB can be reset from the webui (Reset HID, under mouse menu) https://wiki.sipeed.com/hardware/en/kvm/NanoKVM/faq.html#HID-Keyboard-and-Mouse-Not-Working

To downgrade 
https://wiki.sipeed.com/hardware/en/kvm/NanoKVM/faq.html#If-there-are-network-disconnections-or-other-abnormal-situations-during-the-update%2C-it-may-lead-to-a-failed-update.-Please-refer-to-the-following-solutions%3A

mckenm, 2025-03-24:
The HID reset does not work.   May be due to the issue start after the system is either in sleep mode, the HID interface remaiin fail after it reset, when I physically on the system, the device manage show the same error.

I finally found a stable way to have it work is by reboot the NanoKVM from Setting->Device->Reboot.  This seems work even while the computer is under sleep mode.

But I do expect this is an issue that need to fix - since I do not have any idea what may be the issue when the system is either sleep, powerdown, or just blank then I am not able to move a mouse or keyboard to wake it up, it looks like its powered down, but then after you power the computer down and up again, the NanoKVM still fail  and because of it the system power cycle and lost continue service to clients where is not desirable.

Am I the only one that has this issue?

Z2Z-GuGu, 2025-03-24:
Hi, @mckenm The recent updates did not modify the underlying logic of the HID section. Could you please tell me the last usable version number? Additionally, if you encounter this issue, you can try using /etc/init.d/S03usbdev stop_start in the web terminal to resolve it. I believe that restarting can fix the HID issue just like this command can.

mckenm, 2025-03-24:
I just try that HID restart a minute ago, when I restart the Windows after it went to sleep.   The HID remain not working during its boot up, I have to reboot the NanoKVM to make it works again.  I only had the NanoKVM for less than a week, its inital Version as on 2.2.1, that version seems working well after the system went into sleep mode and back,  But than I only have it for less than a week with limited tests/trails.

Z2Z-GuGu, 2025-03-24:
Could you provide the model number of the motherboard? I want to see if I can reproduce the issue.

mckenm, 2025-03-24:
I just tried the "/etc/init.d/S03usbdev stop_start" and it works.   

I also just did a few tests and found that it is not the Computer went to sleep that cuases the issue and its randomly happening to it.    I just saw it went fail while I was just holding the KVM device and plug the AUX USB power to it.   I then try that command and KVM works again.  After that I try to plug and unplug the AUX power and sleep a few time it continues to work. 

So I am now not sure what may havce cause this happen.

My MB is a Asus ProArt X870E-Creator WIFI with AMD 7950X under Windows 11.

- just a FYI, the issue happen after I had it upgrade from 2.2.1 to 2.2.4 and install it thorugh its Web interface.

I will try to observe what may be the root cause of the HID fails.

mckenm, 2025-03-28:
I just update the 2.2.5, but the device still have issue that when the system in either sleep or if I RDP to the system for a long while, the HID devices cannot be detected, the RDP machine shows no HID device has error, but when I try to access the KVM I have to reboot the KVM before it shows the remote screen.      

I also now set HID only mode.

So with 2.2.5 it does not show the USB error, but not able to detect by the system.  It actually worst, since it cannot even see remote screen.

Z2Z-GuGu, 2025-04-01:
Can the system be awakened using the keyboard when it is in sleep mode? If the mouse is in absolute position mode, NanoKVM cannot send mouse information because it cannot obtain the screen size of the sleeping host. Therefore, the only way to wake it up at this time is through the keyboard.

pfre, 2025-04-24:
I received my nanoKVM PCIe early this week, and I’m having the exact same problem. I’ve successfully updated to 2.2.6, and the problem is happening with this version as well. I don’t know if this was happening with my original 2.2.5 version, as I was having HDMI issues and upgraded in an attempt to fix them.

My device’s board is an old MSI H510M-A PRO.

The issue persists with and without HID-only mode. My fix is the same as mckenm’s: to reboot the device. “Reset HID” and “Reset HDMI” don’t have any visible effect.

I’m seeing this right after starting the machine. I’m yet to be sure, but so far it seems that if the first thing I do on the Windows login screen is to use the mouse, everything works. But if I start with the keyboard, sometimes both HID devices stop being recognized by Windows and it eventually takes the screen to sleep.

pfre, 2025-04-25:
Trying to run /etc/init.d/S03usbdev stop_start, I get the errors that I’m attaching. It does not resolve the isuue. I need to reboot the KVM

[image]

prairiemarksman-pixel, 2025-12-11:
I observed something similar but was not able to link it with sleep mode of the host.
Just went into usb root hubs and host controllers, switching off "allow to save power", maybe this will help to at least lower the rate of silently dying HIDs..