sipeed/NanoKVM #405: Not able to access webui after clean install

gregorjan, 2025-03-23:
I downloaded latest firmware version 1.4.0 from [here](https://github.com/sipeed/NanoKVM/releases/tag/v1.4.0) then I flashed the image using ether followind docs [here](https://wiki.sipeed.com/hardware/en/kvm/NanoKVM/system/flashing.html).

After plugging everything in I am getting unable to connect in web browser and getting connection refused while trying to use SSH.

So I tried to get in [USB RNDIS](https://wiki.sipeed.com/hardware/en/kvm/NanoKVM/system/updating.html#Check-via-USB-RNDIS-Network-Interface) however I am getting different address range not starting with 10 

[image]

I tried to ping 169.254.178.1 and 169.254.0.1 with no luck.

So how can I get USB RNDIS working or can I use something else to connect to device?
Also why is ssh disabled by default?

UWH-David, 2025-03-26:
169.254 IP addresses are assigned by Windows boxes indicating there is a network issue with the device. Seems you may not have an accessible DHCP server on your network?

Z2Z-GuGu, 2025-04-01:
Many people are unhappy with the default behavior of having SSH enabled, so in version 1.4.0, I have disabled SSH by default. If you want to use SSH directly after flashing the card, please delete etc/kvm/stop_ssh in the root partition of the TF card.