sipeed/NanoKVM #410: Keyboard not detected by BIOS/EFI during boot sequence

kano-beast, 2025-03-25:
My nanoKVM full (image version 2024-08-17-18-13-713161.img, app version 2.2.4) is able to emulate virtual keyboard once the OS (Windows or FreeBSD) is started, but while in the POST sequence it doesn’t work: I can’t enter in the BIOS menu as example.
Tested on:

HP EliteDesk
Dell Latitude
Lenovo thinkpad e14

Z2Z-GuGu, 2025-03-26:
Hello! We just updated to the latest version 2.2.5, and you should now receive the version update. In this latest version, we added a "HID Only" mode, which enhances HID compatibility and fixes the issue with absolute mouse functionality mentioned in the previous script (of course, some BIOS do not support absolute position mice and can only enable relative mice). Additionally, the previous versions may have had unstable HID performance on some host systems (like the Mac mini M4), and using the "HID Only" mode can fix that as well.

However, the "HID Only" mode will disable the virtual USB drive and virtual network card. If you do not need these additional features, you can prioritize using "HID Only."

You can select the "HID Only" mode under the mouse icon. To ensure a successful switch, the system will restart during the transition. After about 16-23 seconds, you can refresh the webpage automatically or manually, and you will successfully switch to "HID Only" mode.

I tested this on my HP EliteDesk 800 G3, and this mode can reboot smoothly into the BIOS, allowing control with the keyboard and relative mouse.

kano-beast, 2025-03-26:
Hello, It still doesn't work on the Dell and Lenovo laptop. I can't put the password on the bitlocker that comes before the bios

kano-beast, 2025-03-31:
Everything ok now. Thank you.