sipeed/NanoKVM #411: Feature Request - Add ddclient

ckotte, 2025-03-26:
Would be nice if you could add ddclient or something similar to update cloudflare DNS entries. My server is fully encrypted and I want to use NanoKVM to enter the password. If there's a power outage and the UPS doesn't work then I lose internet access to NanoKVM. My router can't update cloudflare.

jaycenornin, 2025-04-12:
Most home routers come with native dynamic dns support. For example, my ASUS router provides dynamic DNS for <mychoice>.asuscomm.com. This is also increasingly common in business class devices, such as FortiGate firewalls.
You can create a CNAME record in your DNS host for whatever domain you want to use (ie cloudflare if you're a glutton for punishment). 

I don't think ddclient would be hard to implement, you could probably install it yourself via SSH with a little elbow grease. Finding riscv64 binaries for the tools you want is a royal pain, and the limited cpu/memory resources on the NanoKVM will be huge bottleneck.

To wit: I got OpenJDK JRE and ScreenConnect to run on mine with the following steps:
- Create a swap partition (prevents OOM crashes)
- Download tar.gz binaries for JDK and extract to /opt
- Copy ScreenConnect binaries (.jar and .so) from a working Ubuntu x64 build
- Update /etc/profile with additional path parameters for OpenJDK
- Some hacking of ScreenConnect startup script to bypass syntax errors between bash and busybox implementations of some commands 

Personally, I don't think ddclient adds significantly to the core functionality of NanoKVM, so _I_ wouldn't prioritize it. But I'm not the dev lol. But you might be able to find a risc binary for perl and get it to run manually. There is an official Ubuntu build for the LicheeRV Nano board that NanoKVM is built on, and that distro may have perl/ddclient and other tools you can use.

ckotte, 2026-01-07:
My router here in Thailand only supports a few (proprietary) DDNS providers and I can't implement DNS updates to Cloudflare with this. However, I could configure this router in bridge mode and use a router from ASUS or another brand and maybe they support Cloudflare.

Now I use ddclient on my server where the NanoKVM is connected to. The only problem with this is if it crashes and hangs in boot prompt then the DNS might time out and I can't connect to my KVM. I use tailscate for this case.

So, this feature request can be closed..