sipeed/NanoKVM #413: No GUI after 2.2.5

Sepzilla, 2025-03-27:
2.2.5 broke the interface and I've rebooted the nanokvm pcie multiple times.

less /var/log/messages

Mar 26 21:29:28 kvm-d62f auth.err getty[854]: can't open '/dev/ttyGS0': No such file or directory^M
Mar 26 21:29:38 kvm-d62f auth.err getty[893]: can't open '/dev/ttyGS0': No such file or directory^M
Mar 26 21:29:48 kvm-d62f auth.err getty[924]: can't open '/dev/ttyGS0': No such file or directory^M
Mar 26 21:29:58 kvm-d62f daemon.info init: Id "acm" respawning too fast: disabled for 5 minutes
Mar 26 21:30:06 kvm-d62f kern.info kernel: [  120.906445] nf_conntrack: default automatic helper assignm
ent has been turned off for security reasons and CT-based  firewall rule not found. Use the iptables CT 
target to attach helpers instead.
Mar 26 21:31:51 kvm-d62f auth.info sshd[1354]: Failed password for root from 192.168.1.243 port 53173 ss
h2
Mar 26 21:31:58 kvm-d62f auth.info sshd[1354]: Accepted password for root from 192.168.1.243 port 53173 
ssh2
Mar 26 21:33:34 kvm-d62f auth.info sshd[1354]: Received disconnect from 192.168.1.243 port 53173:11: dis
connected by user
Mar 26 21:33:34 kvm-d62f auth.info sshd[1354]: Disconnected from user root 192.168.1.243 port 53173
Mar 26 21:33:45 kvm-d62f auth.info sshd[1755]: Accepted password for root from 192.168.1.243 port 53174 
ssh2
Mar 26 21:34:26 kvm-d62f daemon.info ntpd[467]: kernel reports TIME_ERROR: 0x41: Clock Unsynchronized
Mar 26 21:34:59 kvm-d62f auth.err getty[2053]: can't open '/dev/ttyGS0': No such file or directory^M
Mar 26 21:35:09 kvm-d62f auth.err getty[2089]: can't open '/dev/ttyGS0': No such file or directory^M
Mar 26 21:35:19 kvm-d62f auth.err getty[2126]: can't open '/dev/ttyGS0': No such file or directory^M
Mar 26 21:35:29 kvm-d62f auth.err getty[2166]: can't open '/dev/ttyGS0': No such file or directory^M
Mar 26 21:35:39 kvm-d62f auth.err getty[2201]: can't open '/dev/ttyGS0': No such file or directory^M
Mar 26 21:35:49 kvm-d62f auth.err getty[2235]: can't open '/dev/ttyGS0': No such file or directory^M
Mar 26 21:35:59 kvm-d62f auth.err getty[2271]: can't open '/dev/ttyGS0': No such file or directory^M
Mar 26 21:36:09 kvm-d62f auth.err getty[2311]: can't open '/dev/ttyGS0': No such file or directory^M
Mar 26 21:36:19 kvm-d62f auth.err getty[2346]: can't open '/dev/ttyGS0': No such file or directory^M

Z2Z-GuGu, 2025-04-01:
Hello, may I ask if the GUI you mentioned refers to the UI on the web or the UI on the OLED? If the web page doesn't open, please follow these steps to enable the runtime log:

Modify /etc/kvm/server.yaml -> logger -> level: debug
Execute /etc/init.d/S95nanokvm restart to restart the NanoKVM service.

byrevx, 2025-04-12:
Same problem ... is about UI on the web !

I had to move the card to a location without a PCIe slot, because the software reboot didn't solve anything... so it remained powered only by USB.

After a power off-on the interface came back, but this is the second time it happened, the first time it came back only with "reboot -f" via remote ssh, but this time it didn't!

* By the way, another nanovmm pcie interface, newly connected today, got stuck after I configured (initialization) the user/password. I moved it to a location without a PCIe slot and powered it via USB and was able to access the web interface.

I don't know if the problem is related to the PCIe power or not, maybe it's just a coincidence!

# /etc/init.d/S95nanokvm restart
PCIe
oled exist
192.168.100.105
rm: can't remove '/etc/kvm/wifi_exist': No such file or directory
OK
# [SAMPLE_COMM_SNS_ParseIni]-2204: Parse /mnt/data/sensor_cfg.ini
[parse_source_devnum]-1797: devNum =  1
[parse_sensor_name]-1878: sensor =  LONTIUM_LT6911_2M_60FPS_8BIT
[parse_sensor_busid]-1907: bus_id =  4
[parse_sensor_i2caddr]-1918: sns_i2c_addr =  ff
[parse_sensor_mipidev]-1929: mipi_dev =  0
[parse_sensor_laneid]-1940: Lane_id =  2, 4, 3, 1, 0
[parse_sensor_pnswap]-1951: pn_swap =  0, 0, 0, 0, 0
[parse_sensor_mclken]-1973: mclk_en =  0
[parse_sensor_mclk]-1984: mclk =  0
try release sys ok
[_mmf_init]-871: maix multi-media version:67a90e2-musl_riscv64
[SAMPLE_COMM_SNS_ParseIni]-2204: Parse /mnt/data/sensor_cfg.ini
[parse_source_devnum]-1797: devNum =  1
[parse_sensor_name]-1878: sensor =  LONTIUM_LT6911_2M_60FPS_8BIT
[parse_sensor_busid]-1907: bus_id =  4
[parse_sensor_i2caddr]-1918: sns_i2c_addr =  ff
[parse_sensor_mipidev]-1929: mipi_dev =  0
[parse_sensor_laneid]-1940: Lane_id =  2, 4, 3, 1, 0
[parse_sensor_pnswap]-1951: pn_swap =  0, 0, 0, 0, 0
[parse_sensor_mclken]-1973: mclk_en =  0
[parse_sensor_mclk]-1984: mclk =  0
pstSize->u32Width = 1920
pstSize->u32Height = 1080
pstSize->u32Width = 1920
pstSize->u32Height = 1080
pstSize->u32Width = 1920
pstSize->u32Height = 1080
ISP Vipipe(0) Allocate pa(0x8bef4000) va(0x0x3fcc077000) size(291424)
pstSize->u32Width = 1920
pstSize->u32Height = 1080
stSnsrMode.u16Width 1920 stSnsrMode.u16Height 1080 60.000000 wdrMode 0 pstSnsObj 0x3fccdf3e20
pstSize->u32Width = 1920
pstSize->u32Height = 1080
pstSize->u32Width = 1920
pstSize->u32Height = 1080
[SAMPLE_COMM_VI_StartMIPI]-494: sensor 0 stDevAttr.devno 0
[sensor_prob = lt6911_probe] 
###########   PinMux  #######################################################################
###########   PinMux End  #######################################################################
[lt6911_probe] -189 
[lt6911_probe] -195 
[lt6911_i2c_write] : addr = ee. data = 1 
[lt6911_i2c_read] : addr = 0 
[lt6911_i2c_write] : addr = ee. data = 0 
[lt6911_i2c_write] : addr = ee. data = 1 
[lt6911_i2c_read] : addr = 1 
[lt6911_i2c_write] : addr = ee. data = 0 
[lt6911_probe] jump return nVal -198 
[lt6911_probe] -201 
data:ffffffff ffffffff
[lt6911_probe] jump return CVI_FAILURE -206 
[lt6911_probe] -208 = exit success 
pstSize->u32Width = 1920
[... 139 lines cut ...]
[2025-04-12 20:33:01.719] [debug] [router.go:16] router init done
2025/04/12 20:33:01 proto: http, port: 80 443
[hdmi]HDMI-UE exist!
[kvmv] hdmi mode = 1
[kvmv] Cannot obtain HDMI input
[2025-04-12 20:33:03.209] [debug] [jwt.go:66] parse jwt error: token signature is invalid: signature is invalid
[2025-04-12 20:33:03.241] [debug] [jwt.go:66] parse jwt error: token signature is invalid: signature is invalid
[kvmv] VI subsystem is normal
[kvmv] kvmv_cfg.vi_detect_state == 2
[kvmv] kvmv_cfg.vi_detect_state == 2
[kvmv] kvmv_cfg.vi_detect_state == 2
[kvmv] kvmv_cfg.vi_detect_state == 2
[kvmv] kvmv_cfg.vi_detect_state == 2
[kvmv] kvmv_cfg.vi_detect_state == 2
[kvmv] kvmv_cfg.vi_detect_state == 2
[kvmv] kvmv_cfg.vi_detect_state == 2
[kvmv] kvmv_cfg.vi_detect_state == 2
[kvmv] kvmv_cfg.vi_detect_state == 2
[kvmv] kvmv_cfg.vi_detect_state == 2
[kvmv] kvmv_cfg.vi_detect_state == 2
[kvmv] kvmv_cfg.vi_detect_state == 2
[kvmv] kvmv_cfg.vi_detect_state == 2
[kvmv] kvmv_cfg.vi_detect_state == 2
[kvmv] kvmv_cfg.vi_detect_state == 2
... same line endless, web interface not working...