sipeed/NanoKVM #417: [Feature request] Add support for language-specific on-screen keyboard layout

pekishev, 2025-03-27:
It would be nice to have additional language-specific layouts in on-screen keyboard.
For example, when I select Russian display language, I would like to have additional keyboard layout.
It should send same key codes, but display russian letters.

Currently I implemented it myself in forked web, but I had to hardcode it. It would be nice to be language-dependent. 
[image]

pekishev, 2025-04-22:
Created PR #472 to implement this feature.

dormancygrace, 2026-09-14:
Hi! [NanoKVM OS](https://github.com/dormancygrace/NanoKVM-OS), my independent community firmware for SG2002-based NanoKVM Cube and PCIe, includes functionality relevant to this request.

The on-screen keyboard has selectable English, French, German, Russian, Korean and Japanese layouts, including Russian key labels. The selection is saved in the browser.

The layout is selected explicitly in the keyboard, rather than automatically following the Web UI language. It sends HID keys and does not switch the target OS language for you. This includes functionality also developed upstream.

[Implementation 1](https://github.com/dormancygrace/NanoKVM-OS/blob/9f65f9554142db4a61ba866b4498363f4f21d711/web/src/pages/desktop/virtual-keyboard/index.tsx#L47).

If this addresses your use case, you're welcome to try it and let me know how it works on your setup.

It's currently in beta. Development testing has been on PCIe/UXC; Cube testing is still pending. Installing from stock firmware requires flashing a full SD image, so please back up your card first.

[Latest release](https://github.com/dormancygrace/NanoKVM-OS/releases/latest) · [Installation guide](https://github.com/dormancygrace/NanoKVM-OS/blob/main/docs/INSTALL.md)

Please include your board revision and reproduction steps when reporting results. Thanks!