sipeed/NanoKVM #420: Keybord strokes translating to wrong keys on host

Gfiles, 2025-03-28:
When i type certian caracters onthe host it is writing something else.

Examples:
'/' --> 'q'

When i use the virtual keyboard:
'/' --> ';'

Trhought the past function:

"https://www.google.com/" --> "httpsÇ;;www.google.com;"

My keyboard is Brazil ABTN

For the keys to work properly I have to change my keyboard to US. but the paste function still has the same problem

Hoernchen, 2025-03-30:
It is a well known problem see https://github.com/sipeed/NanoKVM/issues/335 and https://github.com/sipeed/NanoKVM/issues/272 and https://github.com/sipeed/NanoKVM/issues/254 and https://github.com/sipeed/NanoKVM/issues/228 the developers don't really care that the device is basically useless so it's time for a refund.

Z2Z-GuGu, 2025-04-01:
Hello, this issue is caused by the host's keyboard settings. Please modify the keyboard type in the host system's settings. I once encountered a problem with a French keyboard; on my computer (Ubuntu), I was able to resolve it by adding French in Settings -> Keyboard -> Input Sources and Input Methods.

funkyfunked, 2025-04-04:
> Hello, this issue is caused by the host's keyboard settings. Please modify the keyboard type in the host system's settings. I once encountered a problem with a French keyboard; on my computer [...]

This did not work for me with ubuntu server. I have a german keyboard and set the host to german keyboard, but the translation is wrong, even if pasting something with the web UI function. Also the virtual keyboard is translating the wrong keys. Tried Chrome, Firefox and even Edge on different clients. 

The only soltion that is working for me is to set the keyboard on the host to english (US). Some keys will be translated wrong with a german (or other) keyboard, but at least | and other special keys are able to be sent. The paste function is also working if the host keyboard is set to "us".