sipeed/NanoKVM #423: Black screen on NanoKVM on latest firmware

JMHershey125, 2025-03-29:
I am recieveing no display from the web UI. I am getting the following error message in the SSH terminal:
[1970-01-02 03:35:30.957] [error] [cgo.go:85] failed to read kvm image: -1

What other information do you need to start troubleshooting?

RomanIrt, 2025-03-29:
I've got the same issue. Reboot doesn't help.

JMHershey125, 2025-03-29:
For reference the remote system I am connecting to are systems built with an ASUS Prime B650M-A wifi motherboard with an AMD Ryzen 7900x using the integrated graphics on the CPU. I am using a brand new 3ft HDMI cable and brand new 32GB micro SD card. I have the NanoKVM Lite model so no OLED Screen or case.

JMHershey125, 2025-03-29:
As a test I will try downgrading to firmware 1.3.0 and report back (I am having this issue on the latest 1.4.0 firmware).

JMHershey125, 2025-03-30:
After downgrading to 1.3.0 I am getting no web ui. I have also tried upgrading to the latest via: python /kvmapp/system/update-nanokvm.py

No change in behavior still a black screen when loading the web ui (I get no video output).

starlit-rocketship, 2025-03-31:
> I am recieveing no display from the web UI. I am getting the following error message in the SSH terminal: [1970-01-02 03:35:30.957] [error] [cgo.go:85] failed to read kvm image: -1
> 
> What other information do you need to start troubleshooting?

for me, I get that message when windows is asleep and the HID inputs don't wake it.
I use short power press and it wakes the PC and gives me display back

JMHershey125, 2025-03-31:
Sorry I should have clarified these systems are running Windows Server 2025 and they don't sleep.

JMHershey125, 2025-04-01:
> @JMHershey125 <https://github.com/JMHershey125> Perhaps some other issues
> are causing the image not to be received. Please execute the following
> commands in the web terminal:
>
> cat /proc/cvitek/vi_dbg
> cat /proc/mipi-rx
>
> This will help me assess the situation you are encountering.

Here is the output from my system running the latest firmware:

# cat /proc/cvitek/vi_dbg
[VI BE_Dbg_Info]
VIPreBEDoneSts :0x0 VIPreBEDmaIdleStatus :0x0
[VI Post_Dbg_Info]
VIIspTopStatus :0x0
[VI DMA_Dbg_Info]
VIWdma0ErrStatus :0x0 VIWdma0IdleStatus :0x0
VIWdma1ErrStatus :0x0 VIWdma1IdleStatus :0x0
VIRdmaErrStatus :0x0 VIRdmaIdleStatus :0x0
[VI ISP_PIPE_A FE_Dbg_Info]
VIPreFERawDbgSts :0x0 VIPreFEDbgInfo :0x0
[VI ISP_PIPE_A]
VIOutImgWidth :1920
VIOutImgHeight :1080
VIInImgWidth :1920
VIInImgHeight :1080
VIDevFPS : 0
VIFPS : 0
VISofCh0Cnt : 0
VIPreFECh0Cnt : 0
VIPreBECh0Cnt : 0
VIPostCnt : 0
VIDropCnt : 0
VIDumpCnt : 0
[VI ISP_PIPE_A Csi_Dbg_Info]
VICsiIntStatus0 :0x0
VICsiIntStatus1 :0x0
VICsiCh0Dbg :0x0
VICsiCh1Dbg :0x0
VICsiOverFlowCnt : 0
VICsiCh0WidthGTCnt : 0
VICsiCh0WidthLSCnt : 0
VICsiCh0HeightGTCnt : 0
VICsiCh0HeightLSCnt : 0
# cat /proc/mipi-rx

Module: [MIPI_RX], Build Time[#30 PREEMPT Mon Nov 11 10:36:06 CST 2024]

------------Combo DEV ATTR--------------
   Devno WorkMode DataType WDRMode LinkId PN Swap Sy
ncMode DataEndian SyncCodeEndian
       0 MIPIYUV422_8BIT NONE 2, 4, 3, 1, 0 0, 0, 0, 0, 0
    N/A N/A N/A

------------MIPI info-------------------
 Devno EccErr CrcErr HdrErr WcErr fifofull decode
     0 0 0 0 0 0 unknown
  Physical: D0 D1 D2 D3 D4 D5
                   0 0 0 0 0 0
   Digital: D0 D1 D2 D3 CK_HS CK_ULPS CK_ST
OP CK_ERR Deskew
             hs_idle hs_idle hs_idle hs_idle 0 0
 0 0 idle
#

Z2Z-GuGu, 2025-04-01:
@JMHershey125 When both VIDevFPS and VIFPS are 0, it is considered that there is indeed no signal on the HDMI. Additionally, if all Physical: D0 D1 D2 D3 D4 D5 of mipi_rx are also 0, this confirms that HDMI has not received a signal, or the HDMI cable may be damaged. You can try connecting any device with an HDMI output. If you replace the cable and still get similar results across all HDMI devices, it is possible that the HDMI receiver on the NanoKVM is damaged. Please contact customer service or email [kvm@sipeed.com](mailto:kvm@sipeed.com) to explain the situation for after-sales service.

JMHershey125, 2025-04-01:
So after plugging in an HDMI monitor and rebooting I was able to get the screen to appear on the monitor. I then moved the HDMI over to the kvm and it works now. Super wierd but thanks for the support. I also noticed that the settings in Windows Server 2025 by default had the monitor sleep after 10 minutes so I turned that to "never" sleep the monitor to make sure the KVM is always accessible.

closed 2025-04-01 by JMHershey125

Z2Z-GuGu, 2025-04-02:
@hendkai It seems that your NanoKVM has actually captured the image, but it is not displaying for other reasons. Please follow these steps to troubleshoot:

1. Check if the version is the latest, 2.2.5.
2. Try restarting to resolve any unknown issues.
3. If the problem persists, open the run log: modify `/etc/kvm/server.yaml`, change `logger->level->info` to `debug`; then execute the command `/etc/init.d/S95nanokvm restart`. Finally, copy the debug information here, and I'll help you analyze it.