sipeed/NanoKVM #424: Compile Issues

Lucasmellof, 2025-03-30:
I can’t compile the server. When I run the commands from the README.md, I encounter this error:

```
root@fedora:/home/system/nanokvm/server# CGO_ENABLED=1 GOOS=linux GOARCH=riscv64 CC=riscv64-unknown-linux-musl-gcc CGO_CFLAGS="-mcpu=c906fdv -march=rv64imafdcv0p7xthead -mcmodel=medany -mabi=lp64d" go build
# NanoKVM-Server
/home/system/host-tools/gcc/riscv64-linux-musl-x86_64/bin/../lib/gcc/riscv64-unknown-linux-musl/10.2.0/../../../../riscv64-unknown-linux-musl/bin/ld: warning: libopencv_video.so.409, needed by /home/system/nanokvm/server/dl_lib/libkvm.so, not found (try using -rpath or -rpath-link) 
```

I tried compiling OpenCV for RISC-V 64 and then copying and referencing the resulting libraries, but that only caused additional errors: 

```
# NanoKVM-Server
/opt/go/pkg/tool/linux_amd64/link: running riscv64-unknown-linux-musl-gcc failed: exit status 1
/home/system/host-tools/gcc/riscv64-linux-musl-x86_64/bin/riscv64-unknown-linux-musl-gcc -o $WORK/b001/exe/a.out -Wl,--export-dynamic-symbol=_cgo_panic -Wl,--export-dynamic-symbol=_cgo_topofstack -Wl,--export-dynamic-symbol=crosscall2 -Wl,--compress-debug-sections=zlib /tmp/go-link-3308681481/go.o /tmp/go-link-3308681481/000000.o /tmp/go-link-3308681481/000001.o /tmp/go-link-3308681481/000002.o /tmp/go-link-3308681481/000003.o /tmp/go-link-3308681481/000004.o /tmp/go-link-3308681481/000005.o /tmp/go-link-3308681481/000006.o /tmp/go-link-3308681481/000007.o /tmp/go-link-3308681481/000008.o /tmp/go-link-3308681481/000009.o /tmp/go-link-3308681481/000010.o /tmp/go-link-3308681481/000011.o /tmp/go-link-3308681481/000012.o /tmp/go-link-3308681481/000013.o /tmp/go-link-3308681481/000014.o /tmp/go-link-3308681481/000015.o /tmp/go-link-3308681481/000016.o /tmp/go-link-3308681481/000017.o /tmp/go-link-3308681481/000018.o /tmp/go-link-3308681481/000019.o /tmp/go-link-3308681481/000020.o /tmp/go-link-3308681481/000021.o /tmp/go-link-3308681481/000022.o -L/home/system/nanokvm/server/dl_lib -Wl,-rpath=/home/system/nanokvm/server/dl_lib -Wl,-rpath-link,/home/system/nanokvm/server/dl_lib -L/home/system/nanokvm/server/dl_lib -lkvm -L/home/system/nanokvm/server/dl_lib -Wl,-rpath=/home/system/nanokvm/server/dl_lib -Wl,-rpath-link,/home/system/nanokvm/server/dl_lib -lpthread -L/home/system/nanokvm/server/dl_lib -Wl,-rpath=/home/system/nanokvm/server/dl_lib -Wl,-rpath-link,/home/system/nanokvm/server/dl_lib -lresolv -L/home/system/nanokvm/server/dl_lib -Wl,-rpath=/home/system/nanokvm/server/dl_lib -Wl,-rpath-link,/home/system/nanokvm/server/dl_lib -no-pie
/home/system/host-tools/gcc/riscv64-linux-musl-x86_64/bin/../lib/gcc/riscv64-unknown-linux-musl/10.2.0/../../../../riscv64-unknown-linux-musl/bin/ld: warning: libopencv_dnn.so.412, needed by /home/system/nanokvm/server/dl_lib/libopencv_video.so.409, not found (try using -rpath or -rpath-link)
/home/system/host-tools/gcc/riscv64-linux-musl-x86_64/bin/../lib/gcc/riscv64-unknown-linux-musl/10.2.0/../../../../riscv64-unknown-linux-musl/bin/ld: warning: libopencv_calib3d.so.412, needed by /home/system/nanokvm/server/dl_lib/libopencv_video.so.409, not found (try using -rpath or -rpath-link)
/home/system/host-tools/gcc/riscv64-linux-musl-x86_64/bin/../lib/gcc/riscv64-unknown-linux-musl/10.2.0/../../../../riscv64-unknown-linux-musl/bin/ld: warning: libopencv_imgproc.so.412, needed by /home/system/nanokvm/server/dl_lib/libopencv_video.so.409, not found (try using -rpath or -rpath-link)
/home/system/host-tools/gcc/riscv64-linux-musl-x86_64/bin/../lib/gcc/riscv64-unknown-linux-musl/10.2.0/../../../../riscv64-unknown-linux-musl/bin/ld: warning: libopencv_core.so.412, needed by /home/system/nanokvm/server/dl_lib/libopencv_video.so.409, not found (try using -rpath or -rpath-link)
/home/system/host-tools/gcc/riscv64-linux-musl-x86_64/bin/../lib/gcc/riscv64-unknown-linux-musl/10.2.0/../../../../riscv64-unknown-linux-musl/bin/ld: /home/system/nanokvm/server/dl_lib/libopencv_video.so.409: undefined reference to `cv::ocl::internal::isPerformanceCheckBypassed()'
/home/system/host-tools/gcc/riscv64-linux-musl-x86_64/bin/../lib/gcc/riscv64-unknown-linux-musl/10.2.0/../../../../riscv64-unknown-linux-musl/bin/ld: /home/system/nanokvm/server/dl_lib/libopencv_video.so.409: undefined reference to `cv::dnn::dnn4_v20241223::Net::forward(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/home/system/host-tools/gcc/riscv64-linux-musl-x86_64/bin/../lib/gcc/riscv64-unknown-linux-musl/10.2.0/../../../../riscv64-unknown-linux-musl/bin/ld: /home/system/nanokvm/server/dl_lib/libopencv_video.so.409: undefined reference to `cv::dnn::dnn4_v20241223::Net::forward(cv::_OutputArray const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/home/system/host-tools/gcc/riscv64-linux-musl-x86_64/bin/../lib/gcc/riscv64-unknown-linux-musl/10.2.0/../../../../riscv64-unknown-linux-musl/bin/ld: /home/system/nanokvm/server/dl_lib/libopencv_video.so.409: undefined reference to `cv::dnn::dnn4_v20241223::readNetFromCaffe(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/home/system/host-tools/gcc/riscv64-linux-musl-x86_64/bin/../lib/gcc/riscv64-unknown-linux-musl/10.2.0/../../../../riscv64-unknown-linux-musl/bin/ld: /home/system/nanokvm/server/dl_lib/libopencv_video.so.409: undefined reference to `cv::dnn::dnn4_v20241223::Net::getLayerId(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const'
/home/system/host-tools/gcc/riscv64-linux-musl-x86_64/bin/../lib/gcc/riscv64-unknown-linux-musl/10.2.0/../../../../riscv64-unknown-linux-musl/bin/ld: /home/system/nanokvm/server/dl_lib/libopencv_video.so.409: undefined reference to `cv::dnn::dnn4_v20241223::Net::~Net()'
/home/system/host-tools/gcc/riscv64-linux-musl-x86_64/bin/../lib/gcc/riscv64-unknown-linux-musl/10.2.0/../../../../riscv64-unknown-linux-musl/bin/ld: /home/system/nanokvm/server/dl_lib/libopencv_video.so.409: undefined reference to `cv::dnn::dnn4_v20241223::Net::empty() const'
/home/system/host-tools/gcc/riscv64-linux-musl-x86_64/bin/../lib/gcc/riscv64-unknown-linux-musl/10.2.0/../../../../riscv64-unknown-linux-musl/bin/ld: /home/system/nanokvm/server/dl_lib/libopencv_video.so.409: undefined reference to `cv::dnn::dnn4_v20241223::Net::setParam(int, int, cv::Mat const&)'
/home/system/host-tools/gcc/riscv64-linux-musl-x86_64/bin/../lib/gcc/riscv64-unknown-linux-musl/10.2.0/../../../../riscv64-unknown-linux-musl/bin/ld: /home/system/nanokvm/server/dl_lib/libopencv_video.so.409: undefined reference to `cv::dnn::dnn4_v20241223::blobFromImage(cv::_InputArray const&, cv::_OutputArray const&, double, cv::Size_<int> const&, cv::Scalar_<double> const&, bool, bool, int)'
/home/system/host-tools/gcc/riscv64-linux-musl-x86_64/bin/../lib/gcc/riscv64-unknown-linux-musl/10.2.0/../../../../riscv64-unknown-linux-musl/bin/ld: /home/system/nanokvm/server/dl_lib/libopencv_video.so.409: undefined reference to `cv::ocl::isOpenCLActivated()'
/home/system/host-tools/gcc/riscv64-linux-musl-x86_64/bin/../lib/gcc/riscv64-unknown-linux-musl/10.2.0/../../../../riscv64-unknown-linux-musl/bin/ld: /home/system/nanokvm/server/dl_lib/libopencv_video.so.409: undefined reference to `cv::dnn::dnn4_v20241223::Net::setInput(cv::_InputArray const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, double, cv::Scalar_<double> const&)'
/home/system/host-tools/gcc/riscv64-linux-musl-x86_64/bin/../lib/gcc/riscv64-unknown-linux-musl/10.2.0/../../../../riscv64-unknown-linux-musl/bin/ld: /home/system/nanokvm/server/dl_lib/libopencv_video.so.409: undefined reference to `cv::dnn::dnn4_v20241223::Net::forward(cv::_OutputArray const&, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&)'
/home/system/host-tools/gcc/riscv64-linux-musl-x86_64/bin/../lib/gcc/riscv64-unknown-linux-musl/10.2.0/../../../../riscv64-unknown-linux-musl/bin/ld: /home/system/nanokvm/server/dl_lib/libopencv_video.so.409: undefined reference to `cv::estimateAffinePartial2D(cv::_InputArray const&, cv::_InputArray const&, cv::_OutputArray const&, int, double, unsigned long, double, unsigned long)'
[... cut ...] obParams()'
/home/system/host-tools/gcc/riscv64-linux-musl-x86_64/bin/../lib/gcc/riscv64-unknown-linux-musl/10.2.0/../../../../riscv64-unknown-linux-musl/bin/ld: /home/system/nanokvm/server/dl_lib/libopencv_video.so.409: undefined reference to `cv::dnn::dnn4_v20241223::Net::setPreferableBackend(int)'
collect2: error: ld returned 1 exit status
```
I’m not sure if I’m missing a specific library or if something else is required for the build to succeed.
A Docker or Nix would be really helpful here.

wj-xiao, 2025-04-03:
Apologies, I missed a step in the README. 

Could you try this step and see if this fixes the problem?

> 3. (Optional) If you compiled libkvm.so yourself, you need to modify its RPATH by `patchelf --add-rpath \$ORIGIN ./dl_lib/libkvm.so`.
> 4. Run `CGO_ENABLED=1 GOOS=linux GOARCH=riscv64 CC=riscv64-unknown-linux-musl-gcc CGO_CFLAGS="-mcpu=c906fdv -march=rv64imafdcv0p7xthead -mcmodel=medany -mabi=lp64d" go build` to compile the project.

Lucasmellof, 2025-04-04:
It didn't work... I still getting the same error:

```
> patchelf --add-rpath \$ORIGIN ./dl_lib/libkvm.so
> CGO_ENABLED=1 GOOS=linux GOARCH=riscv64 CC=riscv64-unknown-linux-musl-gcc CGO_CFLAGS="-mcpu=c906fdv -march=rv64imafdcv0p7xthead -mcmodel=medany -mabi=lp64d" go build               󰚭
# NanoKVM-Server
/home/system/host-tools/gcc/riscv64-linux-musl-x86_64/bin/../lib/gcc/riscv64-unknown-linux-musl/10.2.0/../../../../riscv64-unknown-linux-musl/bin/ld: warning: libopencv_video.so.409, needed by /home/system/temp/nanokvm/server/dl_lib/libkvm.so, not found (try using -rpath or -rpath-link)
```

LingkongSky, 2025-04-04:
I have the same problem too,but the output can work normally.

yanpla, 2025-06-26:
Same issue here