sipeed/NanoKVM #426: 2nd Virtual Disk for BIOS updates/profiles

firedo, 2025-03-30:
Hi, 

It would be nice to have a **2nd smaller Virtual Disk** formatted as **FAT32** for BIOS updates or saving/loading BIOS configuration profiles (backup/restore). The size doesn't have to that big as BIOS update files are usually between 10-50 MiB and BIOS configuration profiles are usually way smaller. Maybe 100-200 MiB's of disk space could be enough for this.

Currently most UEFI BIOSes don't support exFAT format (in the BIOS), so you can't do BIOS updates remotely without physically plugging an USB stick with the BIOS update files. And as FAT32 max. file size is 4GB it's not really feasible to format the current Virtual Disk to FAT32 as most ISO files can be easily over 4GB. I'm assuming that's why exFAT is the default option.

If it's not technically possible to use two Virtual Disks at the same time, there could be an option to switch between the two Virtual Disks.

Thanks!

starlit-rocketship, 2025-03-31:
Once the disk is mounted you can format it.
I even put 2 paritions on it, I used MiniTool Partition Maager Free.
Made one exFAT and one FAT32 and they both mounted.

firedo, 2025-04-02:
Thanks @starlit-rocketship! I actually worked around this by creating an 256 Mb IMG file, that I created and formatted with the ImDisk. I figured from the NanoKVM source code that at least the IMG files are mounted read-write (as long as you don't enable the CD-ROM mode).

I made these simple Scripts to mount & unmount the BIOS update IMG file, so that I could use SFTP/SCP to modify/download the files from inside the IMG file directly on the NanoKVM. It would be kind of slow to download and mount the full IMG file every time you need a small file from inside it.

Upload the IMG file to /data/ folder with the name 'nanoKVM-BIOS-updates-profiles.img' (or you can use a different name, but you have to modify the below scripts as well).

Mount (for ex. "BIOS-updates-profiles-mount.sh"):
`mkdir -p /data/biosupd/
mount -o loop,rw /data/nanoKVM-BIOS-updates-profiles.img /data/biosupd/`

Unmount (for ex. "BIOS-updates-profiles-unmount.sh"):
`umount /data/biosupd/ && rmdir /data/biosupd/`

And then after mounting you can use an SFTP/SCP program to access the files in the /data/biosupd/ folder.

CAUTION: Be sure NOT to write at the same time to the IMG file from the BIOS/guest OS and through SFTP/SCP. Always unmount the IMG file from the NanoKVM menu or with the above Scripts before you mount the other way to be safe. You could corrupt the filesystem by writing at the same time to the mounted IMG file.

To be safer, you can also add a 3rd script that mounts it as read-only by changing the "mount -o loop,**rw**" to "mount -o loop,**ro**".

Sadly the NanoKVM is missing the 'mkfs.vfat' command, so that you could create AND format the IMG file directly on the NanoKVM. But it's no problem as you only need format the IMG file one time.

DrEVILish, 2025-04-05:
It'd be really nice if there was a way to upload an img. Especially for EFI shell configs via the WebUI.

Usecase, I need to open an EFI shell to do some Intel NIC firmware updates. There is no way to upload via an image, or boot into an EFI shell.

fritolays, 2026-09-16:
I landed here from google looking for a means to load a bios file via the nanokvm.

As such I ended up with a shell script, [mkfile.sh](https://github.com/fritolays/notes/tree/main/NanoKVM), that can create an empty but formatted fat32 or exfat image of whatever size. It also allows mounting the image into a local folder so files can be copied into the image.

But like @starlit-rocketship pointed out, once an empty image is created it can be formatted like any usb stick from a computer; ie bootable efi shell.

As a workaround for @DrEVILish, the nanokvm doesnt seem to care if an img file has the iso extension when mounting; which is the only file type that can be uploaded via the webui.

Honestly, I kinda wish it would just allow any file upload.