sipeed/NanoKVM #429: Image distortion macos version 2.2.5

davidnoelopez, 2025-04-01:
Hello, just got my NanoKVM and connected it a mac to test it and I'm seeing image distortion. The mouse and keyboard are working fine. Updated the firmware and tried rebooting mac and KVM but no luck. Any ideas what could it be?

[image]

Z2Z-GuGu, 2025-04-03:
Hello, could you please tell me what the current output video resolution of your Mac is? Would switching it help, for example, changing to 1920×1080 or 1280×720 (not switching in the web page, as the page can only change the transmission resolution)?

davidnoelopez, 2025-04-03:
Hello! I tried switching to all the supported resolutions on the mac and it did the same on all of them.
It is an M2 MacBook Air.
Also tried switching the video mode with no luck :(

Z2Z-GuGu, 2025-04-03:
This situation is quite rare, and I can't confirm if it's a hardware issue. I need you to perform two tests to verify:

1. Try connecting to another HDMI output device.
2. Use a different HDMI cable.

If it still doesn't work, please contact customer service or email kvm@sipeed.com to explain your situation for a return or exchange of the NanoKVM.

davidnoelopez, 2025-04-08:
I tried with another MacBook and it worked. I don't know if it was the macos version that was causing the issue. I don't have the other MacBook Air to test but it had Sonoma IIRC. And the MacBook that worked has Sequoia.
Thanks for your help @Z2Z-GuGu I'll keep testing and hope I don't get any other issues.

closed 2025-04-08 by davidnoelopez