sipeed/NanoKVM #436: 2.2.5 Still Seeing Poor Video and Latency

UWH-David, 2025-04-05:
What I am noticing, even when connected directly over the NIC, is the video quality is pretty low. Whites are grey and the detail is certainly lost. I've tried changing video mode between MJPEG, H.264, lowering resolution, setting quality to lossless, When it comes to the mouse and keyboard, the latency is pretty bad too. This might be great for emergency issues fixing a box remotely but not something I'd want to use daily. Has anyone had any success on improving this situation?

Image Version
v1.4.0
Application Version
2.2.5

Is there client software which I may be missing that would be faster than connecting over the browser web interface by chance? Any VNC backend for example?

Z2Z-GuGu, 2025-04-08:
Hello, could you please tell me what the latency is approximately? Additionally, could you provide your host model?

SergeyKrasiuk, 2025-04-10:
I had a similar situation. Spent a lot of time trying to solve it. I managed to find out something! The problem only occurs on Lenovo mini PCs.
Everything works on other PCs. Regarding the problematic PC, I found one workaround: I connect an additional monitor to it, and the image in nanokvm immediately comes to life. After that, in the screen settings, where the monitor resolution is configured, I select "display only on (nanokvm)", save the setting and everything becomes normal. The second monitor must remain physically connected.
It is worth turning it off and everything freezes or slows down again. I also noticed that the lower the resolution on the second monitor, the less the nanokvm signal delay.

LeonStraathof, 2025-04-18:
What you describe sounds like EDID going wrong and it is corrected by connecting a second monitor. You should be using your NanoKVM without a HDMI switch/extender/splitter in between, so just with the cable that came with it or any other known good normal passive short HDMI cable. If you do that and still have problems it might be that they programmed the EDID phase wrong in NanoKVM as i cannot imagine Lenovo did it wrong on their side.