sipeed/NanoKVM #439: Is my NanoKVM PCIe unit defective?

H0my4k-Naprolom, 2025-04-07:
Hello everyone!
I just got my hands on a recently bought NanoKVM PCIe PoE (non-WiFi version). 
However, it doesn't boot—it just blinks twice with the STA LED after power-on and remains completely silent afterward. 
According to section 1.2 of the FAQ, the unit cannot detect the OS on the flash card.
I’ve tried burning a different image to the flash using both Etcher and dd, and I’ve also tried another flash card—the results are the same. I’ve even tried different power sources: USB from a host, PoE from an Ethernet switch, and PCIe—nothing works.

EnzoAlbornoz, 2025-04-07:
Try unscrewing the PCIe bracket and giving it external power. If the screen lights up after a couple of seconds, your PCIe bracket is pressing the BOOT button and keeping the unit in a boot loop. My unit had this issue. 

If this is confirmed, you can talk with your supplier to get a new PCIe bracket with the correct sizes, or if you have a 3D printer, you can [print a bracket](https://www.printables.com/model/1256646-sipeed-nanokvm-pcie-bracket-full-height).

H0my4k-Naprolom, 2025-04-07:
That works! Thank you very much for your help.

closed 2025-04-07 by H0my4k-Naprolom