sipeed/NanoKVM #443: PCI KVM Dead on Arrival

alooox, 2025-04-09:
Have emailed support and they told me I have to raise an issue here?

My PCI Kvm has arrived dead on arrival, I plugged it into USB power for testing, and nothing happened, tried another PC, nothing happened, tried PCI Power, nothing happened, no lights, no screen etc.

Unsure why this is an issue for GitHub? It’s surely a aftersales product support issue, I’ve already waited months for my PCI KVM to arrive, and now it’s arrived dead. I’d either like a replacement or a refund? 

Purchased direct from the sipeed website.

barrelbarrelbarrel, 2025-04-09:
I had a similar issue with my pcie kvm. Nothing seemed to happen when I plugged in USB power even after trying multiple cables and charging bricks. 

It turns out the 2-pin USB power cable was came inverted. After flipping the connector, it worked perfectly after plugging usb power in again. 

I can't say anything about PCIE power yet.

Z2Z-GuGu, 2025-04-11:
Hello, I suspect you may be experiencing a hardware issue. Although we conduct complete functionality and vibration tests before shipping, we cannot guarantee that the product will be entirely unaffected by shipping impacts. I apologize that you received a faulty product. I will have our customer service team contact you to address the issue.

alooox, 2025-04-14:
When will I be contacted? Still received nothing yet.

S33G, 2025-04-14:
GitHub issues are more for code issues with backend / frontend.
You're likely best to drop an email to support [support@sipeed.com](mailto:support@sipeed.com)