sipeed/NanoKVM #444: Added security solutions to prevent ongoing hacking attempts

sizkein, 2025-04-09:
When nanokvm is connected to the internet, hacking attempts continue to come in. (Login failure)

It would be more secure to leave a login failure history even when a web page login fails.

It would be good to easily install fail2ban or another solution for ssh and web login or to provide it by default.

jaycenornin, 2025-04-11:
NanoKVM, and essentially anything else for that matter, should never be directly connected to the internet. Your firewall should handle intrusion prevention. More importantly, you should be using a VPN or an inside-out remote connection solution to access NanoKVM from a remote network. 

NanoKVM does not have the hardware capabilities to run its own firewall tools. There is simply no way to get fail2ban or snort or other IDS/IPS solutions to run on it. It doesn't have the CPU or RAM to handle much more than the KVM app itself. 

NanoKVM comes with Tailscale and wireguard built-in for inside-out connectivity to your network. If that doesn't work for you, you can stand up a mini-PC head at your remote location and use TeamViewer or some other remote access solution to connect to that PC and use that PC to access your KVM(s).

sizkein, 2025-04-15:
What you said is correct.
However, it is good to have at least a minimal security program.
I think a web login failure history is also necessary.