sipeed/NanoKVM #445: Yubikey

frankozland, 2025-04-09:
Hi - does this device work with yubikey?

also - when presented to the host, what device does the host see the keyboard and mouse? Can that be changed?

jaycenornin, 2025-04-11:
There is currently no support for MFA in this (as far as I can tell). 

The host sees HID compliant input devices for keyboard and mouse.

LeonStraathof, 2025-04-18:
I think you mistake this network KVM with a local USB based KVM. I know the latter can use a Yubikey to forward it to any of your devices. But this should forward the Yubikey trough WebRTC from your endpoint device over the network?? I do not think that is possible without a special software client and special channels for forwarding hardware devices.