sipeed/NanoKVM #448: Feature Request: Stretch image to Window

TheHawkAU, 2025-04-11:
I understand that 1920x1080 is the maximum resolution that is possible, however, the standard behaviour of the video window seems to be to scale down automatically when using a smaller screen/window, but never to scale up beyond the native resolution (which is typically set to 1920x1080 for me).  This means that on 1440 or 4K screens the remote window (even in full screen mode) has giant black borders.

Would it be possible to have a setting to auto scale up the screen to fit the window?

bilibilifmk, 2026-04-28:
https://github.com/sipeed/NanoKVM/pull/795 I fixed this issue and implemented automatic scaling

dormancygrace, 2026-09-14:
Hi! [NanoKVM OS](https://github.com/dormancygrace/NanoKVM-OS), my independent community firmware for SG2002-based NanoKVM Cube and PCIe, includes functionality relevant to this request.

The current viewer fits the video to the available window in both directions, including upscaling beyond the incoming video resolution. The same viewport is used by Direct video, WebRTC and MJPEG.

Aspect ratio is preserved, so borders remain if the window and source have different aspect ratios. This does not add detail to the source image. Similar scaling work has also been discussed upstream in #795.

[Implementation 1](https://github.com/dormancygrace/NanoKVM-OS/blob/9f65f9554142db4a61ba866b4498363f4f21d711/web/src/pages/desktop/screen/viewport.tsx#L68).

If this addresses your use case, you're welcome to try it and let me know how it works on your setup.

It's currently in beta. Development testing has been on PCIe/UXC; Cube testing is still pending. Installing from stock firmware requires flashing a full SD image, so please back up your card first.

[Latest release](https://github.com/dormancygrace/NanoKVM-OS/releases/latest) · [Installation guide](https://github.com/dormancygrace/NanoKVM-OS/blob/main/docs/INSTALL.md)

Please include your board revision and reproduction steps when reporting results. Thanks!