sipeed/NanoKVM #451: I can't update

Andrysky, 2025-04-13:
Image Version v1.3.0
Application Version 2.1.6

[image]

when trying to download the update I get "Get version failed"

[image]

according to the instructions https://github.com/sipeed/NanoKVM/blob/main/server/README.md#manually-update

went to the terminal http://ip/#terminal

downloaded `curl -L https://github.com/sipeed/NanoKVM/releases/download/2.2.5/nanokvm_2.2.5.tar.gz -o nanokvm_2.2.5.tar.gz`

but I can't unzip

```
unzip nanokvm_2.2.5.tar.gz 
Archive:  nanokvm_2.2.5.tar.gz
  End-of-central-directory signature not found.  Either this file is not
  a zipfile, or it constitutes one disk of a multi-part archive.  In the
  latter case the central directory and zipfile comment will be found on
  the last disk(s) of this archive.
unzip:  cannot find zipfile directory in one of nanokvm_2.2.5.tar.gz or
        nanokvm_2.2.5.tar.gz.zip, and cannot find nanokvm_2.2.5.tar.gz.ZIP, period.
```
```
tar -xav -f nanokvm_2.2.5.tar.gz
tar: invalid tar magic
```

the file size looks correct, downloaded several times
```
~ # ls -lah
....
....
-rw-r--r--    1 root     root       14.3M Apr 13 07:53 nanokvm_2.2.5.tar.gz
```

madworm, 2025-04-13:
tar -xzvf something.tar.gz

Andrysky, 2025-04-13:
@madworm 

[image]

dm4web, 2025-04-13:
> @madworm 
> 
> [image]

Try `tar -xf something.tar.gz` or `tar -xaf something.tar.gz`

madworm, 2025-04-13:
If the version of tar apparently support compression/decompression, try this:

`gunzip -c something.tar.gz | tar -xvf -`

Toasterme, 2025-04-14:
or try :

cd /etc/kvm
python ./update-nanokvm.py

Andrysky, 2025-04-20:
```
cd /etc/kvm
python ./update-nanokvm.py
```
didn't help - kvm was no longer available - had to turn off power

`gunzip -c nanokvm_2.2.5.tar.gz | tar -xvf` - helped
but gave
```
~ # /etc/init.d/S95nanokvm restart
killall: kvm_system: no process killed
rm: can't remove '/etc/init.d/S02udisk': No such file or directory
rm: can't remove '/etc/init.d/S30wifi': No such file or directory
rm: can't remove '/etc/init.d/S04backlight': No such file or directory
rm: can't remove '/etc/init.d/S05tp': No such file or directory
```

\+ after that it also became unavailable, turned off the power again and turned it on,

Image Version v1.3.0
Application Version 2.2.5

Image Version - is it updated? (how to update) or does it depend on the PCB?