sipeed/NanoKVM #454: Keyboard issue

coolgoh98, 2025-04-14:
The NANOKVM keyboard has no issue, 
however when i want to use the host keyboard, it always seem to keep holding on CTRL or ALT . 
I not sure how it happen, but i feel when i use alt+tab on the web, the host keyboard will stuck. 

What I can do to avoid having this issue ?
 
Replug the host keyboard does not work. reset HID also does not wrok. 

the softkeyboard at the hose after click it, seem able to clear but still having "Stick ctrl or alt" 

[image]

hanbak, 2025-05-08:
My experience is very similar.
While many people might not encounter this because they use NanoKVM for remote systems in distant locations, I primarily use it to control a company laptop that is right next to me.
This laptop has strict security software installed, and all remote access via network is completely blocked — so NanoKVM is my only option to capture the screen or control it remotely.

However, when I press Alt+Tab on my local computer (the one I'm using to control via NanoKVM), the company laptop often gets stuck with the Alt key held down. I then have to return to the local tab and press Alt again to release it. This is quite frustrating.

Is there any workaround or improvement planned to prevent this kind of key "sticking" issue?

JordiGrasvi, 2025-09-02:
I have the same issue with firmware 2.2.9

rursache, 2025-11-26:
seems like this got fixed in [v2.3.0](https://github.com/sipeed/NanoKVM/releases/tag/2.3.0)

modifier keys are no longer stuck when pressed