sipeed/NanoKVM #455: NanoKVM Lite bricked after update

nightflasher, 2025-04-14:
I did an update some days ago, but after the reboot the web interface didn't came back. So I've waited for about an hour, then I've removed every plug (power/hid, HDMI and ethernet), and connected it to my notebook. There was nothing detected by lsusb und my notebook.

I've flashed the older firmware to the sdcard, which was still in my downloads folder and tried again, but still nothing.

How can I revive my nanokvm lite?

Z2Z-GuGu, 2025-04-15:
May I ask what your old firmware version is? Also, is there a blue LED flashing after powering on?

nightflasher, 2025-04-15:
I don't know about version, I flashed the old 1.4 image and the last working web version was 2.2.4 I guess. but my lite Module had many problems with usb before, had to repin the usb-c because data lines were swapped.

Z2Z-GuGu, 2025-04-16:
The blue LED is used to indicate the system's operating status, while the red LED indicates the power of the NanoKVM. Under normal conditions, the red light stays on continuously, while the blue light will shine briefly upon power-up and then start to flash. Please test by removing the TF card and observing; if the blue LED flashes in a "long on, short off" pattern, you can generally rule out a power failure. Please reflash the [[1.4.0 firmware](https://github.com/sipeed/NanoKVM/releases/tag/v1.4.0)](https://github.com/sipeed/NanoKVM/releases/tag/v1.4.0) and try again. If it still goes out, your NanoKVM may have a hardware fault.

nightflasher, 2025-04-16:
that's the version i was writing to the sdcard.
there seems to be no blue led and i've never seen any blue light on this device

jaycenornin, 2025-04-18:
This link has a pinout diagram
https://wiki.sipeed.com/hardware/en/kvm/NanoKVM/user_guide.html
Pin 16 = UART0 TX
Pin 17 = UART0 RX

I've used this on the Nano Lite, and it is interactive. But if there's a problem causing init to hang it may not be responsive. You will, of course, need your own UART bridge. Apart from losing custom configurations, it may be easier to just reflash the sd card.

nightflasher, 2025-04-29:
NanoKVM defiitely dead - got a new one. working ootb. thx

closed 2025-04-29 by nightflasher