sipeed/NanoKVM #460: 2.2.6 stuck after update

PilaScat, 2025-04-17:
Not returning online, ping doesn't pong
I don't know if I can unplug the usb internal header without turning off the server, obv I cannot turn off the server and in any case it is still powered via PCIe

wj-xiao, 2025-04-18:
Sorry to hear you've run into problems. Please try the following troubleshooting steps:

1. If SSH is enabled on the device, log in via SSH and attempt the update again: [Update](https://wiki.sipeed.com/hardware/en/kvm/NanoKVM/faq.html#Exception-Fixes).
2. Use the `RESET KEY` to perform a device reboot: [image]
3. If the above steps do not resolve the issue, the final option is to reflash the system image: [Flashing](https://wiki.sipeed.com/hardware/en/kvm/NanoKVM/system/flashing.html)

PilaScat, 2025-04-18:
Fixed with reset thank you

closed 2025-04-18 by PilaScat