sipeed/NanoKVM #468: Multiple Monitor Support Not Working

modem-19, 2025-04-19:
Good morning.  I just purchased the NanoKVM PCI-PoE card from Amazon last week and just got it installed this morning.  I have it setup so that it does the full passthrough of the power, reset, and LED's from the front panel to the motherboard headers.  The PC is supplying power via the PCIe slot and I have it on the network successfully.

When I log in to the NanoKVM interface, I get a black screen, even though it indicates it's powered on.  My office PC has 3 monitors connected to an nVidia GeForce 1070ti card with all monitors using the DisplayPort ports and the NanoKVM using the lone HDMI port.  After some troubleshooting it appears that when I unplug all DisplayPort monitors so that I have no active displays, then the NanoKVM shows the screen in the webgui.

This certainly isn't optimal, but a decent work around is that I have remote RMM software on the PC so that the primary function of powering on the PC remotely after a power outage is the main concern.  However, not seeing the BIOS screen during boot to diagnose issues as well as not being able to see the screen in the event there is a crash when my RMM wouldn't work is not the best.

It also appears I have two bad USB A to USB C cables that were shipped with the unit.  I have no Keyboard or Mouse input when the screen is working.

Application Version:  2.2.6
Image Version:  1.3.0

funkyfunked, 2025-04-19:
Whats the resolution on your 3 main displays? Maybe the 1070ti can't handle 4 sumultanious screens with high resolution, even if the connectors are there... Does a normal 4th screen work via hdmi?

modem-19, 2025-04-19:
> Whats the resolution on your 3 main displays? Maybe the 1070ti can't handle 4 sumultanious screens with high resolution, even if the connectors are there... Does a normal 4th screen work via hdmi?

All three original screens are 1080p.  I can plug a 4th monitor into the last DisplayPort, but it doesn't work, but if I plug it into the DVI port, I can get 4 monitors at the same time.

However, I can rule this out because I can plug a single 1080p monitor into the video card and then plug the NanoKVM in, still black screen.

modem-19, 2025-04-19:
A quick update.  It appears once I'm logged into my desktop with 3 monitors, then NanoKVM displays the screen.  However, I'm still left with 2 issues:

1.)  During Boot, POST, BIOS displays, the screen remains black.  As if it's seeing one of the two secondary monitors and not the primary display one.  Again, if I disconnect all monitors and leave only the KVM plugged into the HDMI port, I can see the BIOS screen just fine.

2.)  Icon's don't display on the NanoKVM webgui.  This includes desktop icons, or when I go to interact with the start menu, even the start menu doesn't show on the webgui, even though my PC beside me indicates I indeed did click on it and it opens up on the client.  Just not showing in the webgui.

I did enable HID only mode, but that didn't seem to help at all.

Z2Z-GuGu, 2025-04-22:
> 1.) During Boot, POST, BIOS displays, the screen remains black. As if it's seeing one of the two secondary monitors and not the primary display one. Again, if I disconnect all monitors and leave only the KVM plugged into the HDMI port, I can see the BIOS screen just fine.

@modem-19 Hello, regarding this issue, NanoKVM only simulates a regular display, but whether it is used as the primary display is controlled by the system. There may also be related settings in the BIOS, such as options like "Boot Display" or "Primary Display."

The USB keyboard and mouse issue: could you please let us know if the USB icon on the OLED lights up when in use? Additionally, could you provide us with the model number of your motherboard?

Captain-Pumpkinhead, 2025-04-24:
I've come across an interesting multi-monitor issue.  Even when the KVM is set as Monitor \#2, it only moves the mouse on monitor \#1.

I hope this gets fixed.

Z2Z-GuGu, 2025-04-25:
> I've come across an interesting multi-monitor issue. Even when the KVM is set as Monitor # 2, it only moves the mouse on monitor # 1.
> 
> I hope this gets fixed.

You might try using relative mouse mode, as the default absolute position mouse works best with a single display.