sipeed/NanoKVM #474: Add option to initiate a clean shutdown of a Linux system with the SysRq + REISUB combination

firedo, 2025-04-23:
Please, add a shortcut in the NanoKVM menu to press the keys Alt + Sysrq and then while still holding down Alt press these keys in succession:
R, E, I (big i), S, U and B

This will initiate a clean shutdown of an frozen Linux operating system, as long as the feature has been enabled.

I tried to find out how to implement this with the Scripts, but I couldn't figure out how to keep the Alt key pressed down while pressing the other keys down.

nickshanks347, 2025-05-05:
What did you come up with for the scripts?

droopanu, 2025-09-19:
I managed to create this script to sent sysrq+s (sync). You can adapt it to send any other combinations:

`echo -ne \\x04\\x00\\x46\\x00\\x00\\x00\\x00\\x00 > /dev/hidg0  #Pressing Alt + SysRq
sleep 0.1                                                                                     
echo -ne \\x04\\x00\\x46\\x16\\x00\\x00\\x00\\x00 > /dev/hidg0  #Pressing s while holding Alt + SysRq
sleep 0.1                                                           
echo -ne \\x04\\x00\\x46\\x00\\x00\\x00\\x00\\x00 > /dev/hidg0  #Releasing s
sleep 0.1                           
echo -ne \\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00 > /dev/hidg0  #Releasing all keys

`

For other characters replace the x16 to the desired one:
r 0x0015
e 0x0008
i 0x000C (not sure about capital I tho)
u 0x0018
b 0x0005