sipeed/NanoKVM #479: RDP/VNC server?

raingloom, 2025-04-25:
The browser interface is nice in theory and it's usually good that users don't have to install anything, but for power users it might be useful to be able to use an RDP or VNC client.  RDP has features like USB and audio passthrough and clients usually support capturing all input events, so there would be no issue with shortcuts like `ctrl-w` going to the browser and closing the viewer tab.

Downside: NanoKVM-specific features like RNDIS or ISO mounting would likely not have equivalents on an RDP/VNC client.  (Although ISO mounting could be done through RDP file sharing.)

Would something like this be feasible / is there interest in it?

jaycenornin, 2025-05-05:
I think this is the same idea as #449 
I'm not a dev on this project, just someone fiddling in my own time, but so far I haven't gotten it to work. But it is feasible.

polyzium, 2025-05-06:
As a temporary workaround you can launch Chrome/Chromium in application mode:
`chromium --app="http://xxx.yyy.zzz.aaa`

raingloom, 2025-05-07:
polyzium ***@***.***> writes:

> As a temporary workaround you can launch Chrome/Chromium in application mode:
> chromium --app="http://xxx.yyy.zzz.aaa

Thanks, unfortunately it doesn't work at all in Ungoogled Chromium.

Funkelfetisch, 2025-05-15:
that would be cool, I think most annoyances come from noVNC!

lowtech-guy, 2026-08-16:
Try this image for nanokvm: https://github.com/onekvm/onekvm-release-nanokvm

Looking forward to feedback