sipeed/NanoKVM #481: Request for feature: capture screenshot of the screen

gdamjan, 2025-04-27:
It'd be nice to be able to capture a screenshot of the host screen, directly from the NanoKVM UI.
Once clicked it can give us the option to copy to the clipboard or download the screenshot as an image file.

The alternatives are:
- to use your OS screenshotter, but that **would not** get the native resolution of the host screen, unless you fit everything just right, if even
- or the browser (firefox) option for screenshots, which is a bit weird, since you need to click on the correct part of the browser, and still then the screenshot image you get is not really synced with the source video frame and you get visible glitches

scpcom, 2025-05-03:
I also think a snapshot button would be useful.
But in Firefox (H.264 mode) you can already use Shift + Right Mouse Click on the middle of the screen and then Select "Take Snapshot..." (not screenshot) to get only the KVM screen picture.

dormancygrace, 2026-09-14:
Hi! This functionality is available in [NanoKVM OS beta-10](https://github.com/dormancygrace/NanoKVM-OS/releases/tag/v1.0.0-beta.10), my independent community firmware for SG2002-based NanoKVM Cube and PCIe.

A camera button in the viewer saves the current frame as PNG at the decoded video dimensions, without the Web UI toolbar. It supports Direct, WebRTC and MJPEG and captures in the browser without writing the screenshot to the NanoKVM SD card.

PNG download is implemented; copying the image to the clipboard is separate. The screenshot uses the stream resolution, which can differ from HDMI input resolution if downscaling is enabled.

[Implementation 1](https://github.com/dormancygrace/NanoKVM-OS/blob/12a1e169727eab73dd335049eb9d67fc427e2035/web/src/pages/desktop/menu/screenshot.tsx) · [Implementation 2](https://github.com/dormancygrace/NanoKVM-OS/blob/12a1e169727eab73dd335049eb9d67fc427e2035/web/src/lib/screenshot.ts).

You're welcome to try it and let me know how it works on your setup. Development testing has been on PCIe/UXC; Cube qualification is still pending. Installing from stock requires flashing the full SD image, so please back up your card first. Existing beta-9 installations with the compatible 64 MiB boot layout can use the signed system update.

[Release and downloads](https://github.com/dormancygrace/NanoKVM-OS/releases/tag/v1.0.0-beta.10) · [Installation guide](https://github.com/dormancygrace/NanoKVM-OS/blob/12a1e169727eab73dd335049eb9d67fc427e2035/docs/INSTALL.md)

Please include your board revision, HDMI source, browser, codec and transport when reporting results. Thanks!