sipeed/NanoKVM #484: Mouse cursor movement issue with 16:10 screen aspect ratio

DienoX, 2025-04-28:
Hi,
It occurs mouse cursor movement issue with 16:10 screen aspect ratio.
When using KVM the mouse has an increasing or decreasing offset compared to the host cursor. In the center of the screen the rate is almost perfect. On the left and right the edge is terribly offset between the remote and physical cursor.

Please fix it. I have the latest version of nanokvm software update

yaunz, 2025-05-06:
I have the same problem with 16:10 screen

itsgps, 2025-09-07:
OK - so the issue is - on the 16:10 - you likely have a resolution like 1920 x 1200 rather than 1920 x 1080.  Thing is the NanoKVM really only likely supports standard aspect ratios.

[image]
As such - if you change the laptop to 1920 x 1080 it should be fine - Mine was!

DienoX, 2025-09-07:
This thread is about not changing the resolution

itsgps, 2025-09-07:
> This thread is about not changing the resolution

Actually it doesn't specifically say that - it just notes that it is for 16:10 aspect.  Adding the info I have may assist from a Development perspective - and/or for anyone else having the same issue to have a work around.  i.e. I am more than happy to have my screen at 1920x1080 when using the laptop from my PC - versus having to use the laptop directly - because my PC has a 49" Ultra wide and a 32" monitor attached.  So was always easier for me to RDP to laptop - which at the moment is not working.

SO for me the reset to 16:9 is a work around - but I definitely would like it fixed as I am sure anyone else seeing the thread would also like.

JustSomeGuyNZ, 2025-09-09:
+1 for requesting a fix for this.

Ivan-Firefly, 2025-10-04:
Same for me, 16:10 works fine for vertical axis and has offset on the edges of the screen for horizontal axis

wj-xiao, 2025-11-27:
The issue is caused by an aspect ratio mismatch: the actual display is 16:10, but NanoKVM is outputting a 16:9 stream.
Since the mouse coordinates are calculated based on the 16:9 resolution, this results in mouse drift.

The key is to ensure NanoKVM outputs a 16:10 image. If you are controlling a Windows system, this requires a few extra steps:

1. Set the resolution to **Auto** in the NanoKVM web UI.
2. Manually set the **actual resolution** to 16:10 inside the remote Windows system settings.

Please refer to this [document](https://wiki.sipeed.com/hardware/en/kvm/NanoKVM_Pro/faq.html#HDMI-Icon-Lit-on-NanoKVM-Screen-but-No-Video-Displayed-in-Web-Interface) for more details.