sipeed/NanoKVM #485: No video - v1.4.0/2.2.0 or with the 2.2.6 update

GregE777, 2025-04-29:
I recently received my NanoKVM Lite and when I first took it out of the package, it worked (though the hardcoded DNS wouldn't let me update)

I finally figured out how to fix the DNS issues, and updated to 2.2.6. Then I brought it back to my lab rack and connected it to a computer, there was no video, but mouse and keyboard seemed to work. After a bunch of messing around, I wiped the SD card and flashed 1.4.0 install image again, and brought it back to the computer, and still no video. I tried changing the resolution and codec and frame rate, with rebooting the computer in between changes, still nothing.

I tried sending it some other video sources that I had present, both over Displayport to HDMI and HDMI, one source was not a computer so I know it was stable and was known to be 1920x1080.

Since it worked when I first received it, and it no longer works after flashing back to the 1.4.0 image, I'm guessing it is broken. What's my next step?

msklywenn, 2025-05-08:
I just received mine with firmware 2.2.5. It worked fine. I updated to 2.2.6, no video since.

GregE777, 2025-05-08:
Thanks. Have you tried going back to the 1.4.0. or 1.3.0 card image? I haven't had time to try the 1.3.0 yet, but it still didn't work when I went back to a fresh 1.4.0. I'd find it hard to believe that the 2.2.6 application would kill the video, but maybe.

The worst part is no way to contact Sipeed, at least I can't find one. I want this to work, but right now not going to buy another just to have it fail early again.

xciv, 2025-05-10:
Just as a data point.. I flashed 1.4.0 (it included app version 2.2.0) to my brand new Lite and found it was very very flakey until I managed to do the in system update to 2.2.5 and it is stable since.

The weirdness I observed was that it would get into a state where all the fields (IP address etc) in the About screen were empty.

kuroiookami77, 2025-05-13:
The same problem occurred when upgrading to 2.2.6. I flashed the SD card with version 20250217_NanoKVM_Rev1_4_0 - the video appeared.