sipeed/NanoKVM #486: Any documentation

Softbox44, 2025-05-01:
Is there any documentation on how to use this device. I dont need hardware specs. What we need is an understanding how to use it.
My use case:
I have KVM connected via usb to a server that has no boot drive currently
The server has an empty USB stick attached that I want to use to create a boot drive.
I have download the boot iso image to KMV but I cannot see what I can do with the image, espcially how I can get it borned to the USB stick. I cannot even see the USB stick from the KVM.
Any ideas?

scpcom, 2025-05-03:
The USB drive is simulated by NanoKVM, you do not need a real USB stick attached.
The documentation is here:
https://wiki.sipeed.com/hardware/en/kvm/NanoKVM/user_guide.html#ISO-Image-Mounting-and-Remote-Installation

Softbox44, 2025-05-06:
thanks but the documentation is incomprehensible. How do I select an ISO at boot?
My NanoKBM is stuck showiong a black screen and te word GRUB. No keyboard entry works.

scpcom, 2025-05-09:
> thanks but the documentation is incomprehensible. How do I select an ISO at boot? My NanoKBM is stuck showiong a black screen and te word "RUB loading." No keyboard entry works.

1. Make sure the server is connected to the USB HID port of NanoKVM (if you have NanoKVM Cube it is the USB port below HDMI, the location is shown with a blue square on the sticker)
2. You have to Click the CD-Icon on the NanoKVM menu bar then click on the name of the iso you want to use.
3. Power on the server and press the key (refer to the user manual of your server) to enter the BIOS or the boot menu and select USB CD/DVD-ROM (Linux File Gadget) as boot device

> Maybe I have accidentally erased the NanoKBM image. How can I re-install that?

If you did erase the firmware image:
https://wiki.sipeed.com/hardware/en/kvm/NanoKVM/system/flashing.html