sipeed/NanoKVM #491: PC boot screen - no keyboard

marian790, 2025-05-07:
When PC boot (older PC AMD P2 955 with Asus MOBO BIOS, not UEFI), at post messages cant hit DEL (to enter BIOS) or CTRL+ALT+DEL, same with virtual keyboard.

But when boot to operatin system, keyboard and also virtual keyboard is working.

PCIe nanoKVM
image: v1.4.0
app: 2.2.5

marian790, 2025-05-07:
ok, so updated to 2.2.6 and try on 3rd PC, with bios - keyboard not working

marian790, 2025-05-07:
so, SSH to nanoKVM, created /boot/BIOS, reboot nanoKVM, keyboard on boot screen still not working

marian790, 2025-05-07:
set in BIOS, USB legacy support from auto to enabled, keyboard not work, still not get on bootscreen to hit DEL or other keys...

marian790, 2025-05-11:
another PC - HP 1001 - UEFI AMD R5 4600G - UEFI boot kayboard WORKING!!!

So what is wrong, when on old BIOS PC and newer MSI UEFI mobo not?

Today after testing updated app also to v2.2.7, so i will test again on MSI UEFI and ASISU BIOS mobo.

svatoslove, 2025-05-14:
Me to, keyboard dont work in boot mode, nanoKVM Full, 2.2.6

svatoslove, 2025-05-15:
> another PC - HP M01-F1002nc - UEFI AMD R5 4600G - UEFI boot keyboard WORKING!!!
> 
> So what is wrong, when on old BIOS PC and newer MSI UEFI mobo not?
> 
> Today after testing updated app also to v2.2.7, so i will test again on MSI UEFI and ASISU BIOS mobo.

On some computers, nanoVM didn't work for me via the DP to HDMI adapter in the BIOS, try changing the cable, mb it will help.

svatoslove, 2025-05-15:
As far as I understand, BIOSes are very different, there is a regular old BIOS, there is a regular UEFI, there is a branded UEFI like for example in HP. And everywhere it works differently with the monitor and HID.

jlinton, 2025-07-19:
This is likely because the nanokvm is failing to report a  boot interface subclass.

A more detailed comment: https://github.com/sipeed/NanoKVM/issues/491

Baldoma, 2025-11-05:
I just received the nanoKVM, and I've already encountered the keyboard problem in the BIOS. I noticed that when I restarted the nanoKVM, the keyboard then worked.