sipeed/NanoKVM #492: [Feature Request] - Disable/Enable NanoKVM Display (Toggle)

starlit-rocketship, 2025-05-07:
I have a use case where I have 2 monitors on my desktop, and have the NanoKVM added as well so I can access my PC when I am not home.
This leaves me with a "tripple" montior setup where sometimes apps get stuck on the third 'NanoKVM Monitor'

When I physically leave my desk, I can turn my ACTUAL monitors off. So this is not an issue when using the NanoKVM as intended. (I actually use a smart plug to turn them on/off).

However, this creates an issue where (when in an OS) there is a third monitor showing when I don't need it.

I am sure others probably have a similar scenario too. Where people have multi monitors at times they don't require it. Perhaps on non headless servers.

My request is that we add a toggle to enable/disable the NanoKVM display. 
So the OS will not see the second/third NanoKVM monitor unless toggled on.

Not sure if this is possible? But would help my use case.

Thanks for all the work so far!

tpretz, 2025-05-10:
Did a bunch of debugging, in the end i could only get this functional on the PCIe version of the KVM. 
The product maintainers may know of / be able to add a mechanism that works for other hardware versions that i am not able to. 

https://github.com/sipeed/NanoKVM/pull/501

starlit-rocketship, 2025-05-10:
Thanks for that. 
I actually use the pci-e version across 3 systems so this will work great for me

svatoslove, 2025-05-14:
It would be great if it was possible to disable/enable HDMI connection via the NanoKVM control panel. It was a very useful function for both, PCIe and NanoKVM Full.

wj-xiao, 2025-05-28:
Thanks to @tpretz , this feature is now available in version [2.2.8](https://github.com/sipeed/NanoKVM/blob/main/CHANGELOG.md#228-01e28f1-2025-05-22).
Due to hardware limitations, this update is specific to the PCIe version for now.

closed 2025-05-28 by wj-xiao