sipeed/NanoKVM #495: NO network , ethernet is down

nippit, 2025-05-09:
Background & Target
I Have purchased
1 NanoKVM Lite
and
1 Nano KVM PCIe PoE WiFi

I have etched "20250217_NanoKVM_Rev1_4_0.img.xz" on to the SD cards

plugged in network and power
plugged in network and power and HDMI
pluuged in PoE switch (PoE PCIe version)

tested with unifi managed switches (tried manual 100Mbps HDP and FDP)
tested on a 10/100 only switch

NO switch port lights up.

Expected & Actual
Expected :
-update latest image
-plug in power and network
-network port connected to light up
-find the device on the DHCP table

ACTUAL
no switch port lights up at any condition
(tested with unifi managed switches (tried manual 100Mbps HDP and FDP)
tested on a 10/100 only switch)

Nano KVM PCIe PoE WiFi screen displays ---

Platform
1 NanoKVM Lite
and
1 Nano KVM PCIe PoE WiFi

Problem recurrence
reproduce at will

nippit, 2025-05-09:
NOTE: 
-when I run PCIe PoE model using external 5V usb PSU it is NOT booting. > i use battery pack
-when I connect PCIe PoE model to PoE switch port NOTHING is happening

Z2Z-GuGu, 2025-05-13:
@nippit Hello, I'll try to solve your problem
First, investigate the issue with your PCIe version. Please check the flashing status of the blue STA light on the PCIe panel
If it goes off: it is often a power issue
If it flashes irregularly: start normally
If there is a regular on-off cycle, it is usually a TF card burning error

nippit, 2025-05-13:
@Z2Z-GuGu 
thank you for the repsonse
PCIe version > plug into PoE switch >NOTHING Happesn
PCIe version > Plug into 5V PSU > nothing happens
PCIe version > only power on from battery pack

how can I use this with a battery pack? PoE? PSU?

nippit, 2025-05-13:
PCIe version
power from battery
replace SD card with new one.
power on
blue led blink irregularly (like reading data)
Oled screen lights up
IP address ---
no light on the switch for teh etherenet link up.

what is next suggestion

nippit, 2025-05-15:
I have two of these units both notworking.
No one is replying 
this is very bad support

Tail870, 2025-07-25:
I've created a pull request: https://github.com/sipeed/sipeed_wiki/pull/834

Tail870, 2025-08-07:
@nippit, I wonder if you still having this issue...

Mifody-07, 2026-10-29:
I’m not certain if other versions of the NanoKVM have the same grounding issue, but you should still try grounding both the NanoKVM and the network switch. look at my post in this https://github.com/sipeed/NanoKVM/issues/26#issuecomment-6373769940