sipeed/NanoKVM #499: 640x480 VGA input not working

xciv, 2025-05-10:
Hardware: Lite
Image: 1.4.0
App: 2.2.6

I realise that my use case is unconventional but I can't be the only one wanting to plug into older hardware and get initial POST and BIOS access.

I am using a VGA to HDMI adaptor (BENFEI) and get no video display inside the web interface at power on, once the system has booted further and switched to 1152x864 I see output (although the bottom section of the screen is chopped off, but that's another issue).

(for reference: https://www.amazon.co.uk/dp/B07K14NR8P)

If I plug the output into my HDMI screen the VGA adaptor works and the screen is seeing 640x480 @ 60Hz.

The monitor and HDMI cable (v1.something probably) are rather old, if hardware is relevant.

xciv, 2025-05-11:
I had a poke around for some debugging information:

```
# cat /etc/kvm/hw
beta
# cat /etc/kvm/hdmi_version
ux
# cat /etc/kvm/hdmi_mode
1

```

```
# cat /proc/cvitek/vi_dbg 
[VI BE_Dbg_Info]
VIPreBEDoneSts		:0x27		VIPreBEDmaIdleStatus	:0x7
[VI Post_Dbg_Info]
VIIspTopStatus		:0x3fe
[VI DMA_Dbg_Info]
VIWdma0ErrStatus	:0x3000000	VIWdma0IdleStatus	:0xfffffffe
VIWdma1ErrStatus	:0x3000000	VIWdma1IdleStatus	:0xffffffff
VIRdmaErrStatus		:0x3000000	VIRdmaIdleStatus	:0xffffffff
[VI ISP_PIPE_A FE_Dbg_Info]
VIPreFERawDbgSts	:0x3f0		VIPreFEDbgInfo		:0x3e
[VI ISP_PIPE_A]
VIOutImgWidth		:1152
VIOutImgHeight		: 864
VIInImgWidth		:1152
VIInImgHeight		: 864
VIDevFPS		:   0
VIFPS			:   0
VISofCh0Cnt		:184956
VIPreFECh0Cnt		:219809
VIPreBECh0Cnt		:   0
VIPostCnt		:   0
VIDropCnt		:   0
VIDumpCnt		:   0
[VI ISP_PIPE_A Csi_Dbg_Info]
VICsiIntStatus0		:0x6
VICsiIntStatus1		:0x0
VICsiCh0Dbg		:0x0
VICsiCh1Dbg		:0x0
VICsiOverFlowCnt	:   0
VICsiCh0WidthGTCnt	:   0
VICsiCh0WidthLSCnt	:   0
VICsiCh0HeightGTCnt	:   0
VICsiCh0HeightLSCnt	:   0

```

marian790, 2025-05-21:
Same problem on my setup.

kimnzl, 2025-05-28:
I also had issues using a VGA to HDMI converters.
I ended up working around it by getting a VGA EDID Emulator Passthrough Adapter with an EDID for 1080p..

marian790, 2025-05-28:
> I also had issues using a VGA to HDMI converters. I ended up working around it by getting a VGA EDID Emulator Passthrough Adapter with an EDID for 1080p..

I will try....