sipeed/NanoKVM #505: Issue with capital leters

piotr-siedlak, 2025-05-12:
When i try to login to ESX on intel NUC i can use Capital letters as on screen.
When i Use Shift or CapsLock i aways have first letter small and next big one. 
[image]

piotr-siedlak, 2025-05-12:
HID-only mode disabled and enabled, the same situation.
Changing cables, rebooting etc the same situation.

ajw107, 2025-05-17:
I have a similar issue with the shift key, when holding it down to select text with an arrow key, it does not work.  It does a key-up of the shift key after every character, or sometimes does not register it at all.

piotr-siedlak, 2025-07-23:
newest version still have the same issue.

tefod-zz, 2025-09-01:
Can confirm - have the exact same issue. Haven’t found any setting that could affect this. 
A capital letter only works on the second try when holding SHIFT. It’s not related to the physical keyboard - the same happens with the virtual on-screen keyboard.