sipeed/NanoKVM #506: [Bug] 2.2.7. Authentication is not disabled when authentication: disable is set

arbdevml, 2025-05-12:
Hello.
First of all very big thank you for this project.

**Environment**:
- Device: NanoKVM Full
- Image Version: 1.4.0
- Application Version: 2.2.7
- Web Browser: the last version of Chrome and Firefox

**Steps to Reproduce**:
1. Image 1.4.0 
2. Update manually to 2.2.7
3. Set `authentication: disable` in `/etc/kvm/server.yaml`
4. Restart service: `/etc/init.d/S95nanokvm restart`
5. Access web interface

**Expected Behavior**:
- The web interface should open without the login screen

**Actual Behavior**:
- The login screen still appears

How to disable the login screen?
Thank you for your answer.

ensto1977, 2025-05-29:
Hello,

seems like a docu issue.
I changed authentication to Authentication inside /etc/kvm/server.yaml and it works.

please see: server/service/auth/login.go:   if conf.Authentication == "disable" {

br enrico

arbdevml, 2025-05-30:
@ensto1977 thank you for your answer.

I had manually updated (https://github.com/sipeed/NanoKVM/issues/509), uploaded the tar archive (scp).
nanokvm without internet access.

# rm /etc/kvm/server.yaml

# /etc/init.d/S95nanokvm restart

# cat /etc/kvm/server.yaml 
proto: http
port:
    http: 80
    https: 443
cert:
    crt: server.crt
    key: server.key
logger:
    level: info
    file: stdout
authentication: enable
jwt:
    secretKey: ""
    refreshTokenDuration: 2678400
    revokeTokensOnLogout: true
stun: stun.l.google.com:19302
turn:
    turnAddr: ""
    turnUser: ""
    turnCred: ""
    

modified the /etc/kvm/server.yaml, Authentication: disable

# cat /etc/kvm/server.yaml 
proto: http
port:
    http: 80
    https: 443
cert:
    crt: server.crt
    key: server.key
logger:
    level: info
    file: stdout
Authentication: disable
jwt:
    secretKey: ""
    refreshTokenDuration: 2678400
    revokeTokensOnLogout: true
stun: stun.l.google.com:19302
turn:
    turnAddr: ""
    turnUser: ""
    turnCred: ""

# /etc/init.d/S95nanokvm restart 2>&1 | grep "Authentication"
1970/01/01 00:12:44 NOTICE: Authentication is disabled! Please ensure your service is secure!

# /etc/init.d/S95nanokvm restart 2>&1 | less
 to use /, searching by the word "Authentication"

1970/01/01 00:17:27 NOTICE: Authentication is disabled! Please ensure your service is secure!
1970/01/01 00:17:27 config loaded successfully
[1970-01-01 00:17:27.370] [info] [logger.go:50] logger set success

# go to http://nanokvmip
The login form is still present

@ensto1977 when you visit http://nanokvmip what did you see?

ensto1977, 2025-05-31:
Hi,

i‘m using the https protocol and directly reached the managment webpage. So no login credentials needed. 
I‘m using safari / ipad 

ah:

if i‘m switching to chrome - it‘s are asking  (one time) for login /password - so it seems it‘s needs a cookie to save the login data.

br enrico

ricardomaia, 2025-08-02:
> Tentei todas as combinações que pude com isso, e ele ainda me mostra o formulário de login.
> 
> Se eu remover o cookie do endereço IP do IPKVM, ele sempre solicitará login.

Same here! I'd tried to disable authentication into `server.yaml` file, but still no result.There are any updates about this issue? I've not found any other discussion about it, even in Sipeed Telegram channel.

wurststulle, 2026-08-18:
still having the same problem. anyone got any luck here?

dormancygrace, 2026-09-14:
Hi! [NanoKVM OS](https://github.com/dormancygrace/NanoKVM-OS), my independent community firmware for SG2002-based NanoKVM Cube and PCIe, includes functionality relevant to this request.

With authentication: disable, the middleware supplies an admin principal without requiring a session cookie. The account endpoint returns that principal, and the Web UI uses the account response to enter the desktop rather than requiring a stored login token.

This intentionally gives unauthenticated visitors administrator access. I traced the middleware, account endpoint and frontend route; a fresh-browser test on your device remains to be confirmed.

[Implementation 1](https://github.com/dormancygrace/NanoKVM-OS/blob/9f65f9554142db4a61ba866b4498363f4f21d711/server/middleware/jwt.go#L140) · [Implementation 2](https://github.com/dormancygrace/NanoKVM-OS/blob/9f65f9554142db4a61ba866b4498363f4f21d711/server/service/auth/login.go#L108) · [Implementation 3](https://github.com/dormancygrace/NanoKVM-OS/blob/9f65f9554142db4a61ba866b4498363f4f21d711/web/src/components/auth.tsx#L9).

If this addresses your use case, you're welcome to try it and let me know how it works on your setup.

It's currently in beta. Development testing has been on PCIe/UXC; Cube testing is still pending. Installing from stock firmware requires flashing a full SD image, so please back up your card first.

[Latest release](https://github.com/dormancygrace/NanoKVM-OS/releases/latest) · [Installation guide](https://github.com/dormancygrace/NanoKVM-OS/blob/main/docs/INSTALL.md)

Please include your board revision and reproduction steps when reporting results. Thanks!