sipeed/NanoKVM #509: Install Application Update Without Internet Access

krische, 2025-05-13:
I can't seem to find any documentation on it, but is it possible to update the NanoKVM application without internet access?

I see there are `.tar.gz` files released on GitHub. Am I able to download the release from another computer and transfer it over to the device to install?

mdeanda, 2025-05-14:
Alternatively, it would be nice to be able to provide an alternate download server that can be hosted in internal network.

arbdevml, 2025-05-14:
@krische
https://github.com/sipeed/NanoKVM/tree/main/server#manually-update
## Manually Update

> File uploads requires SSH. Please enable it in the Web Settings: `Settings > SSH`;

1. Download the latest application from [GitHub](https://github.com/sipeed/NanoKVM/releases);
2. Unzip the downloaded file and rename the unzipped folder to `kvmapp`;
3. Back up the existing `/kvmapp` directory on your NanoKVM, then replace it with the new `kvmapp` folder;
4. Run `/etc/init.d/S95nanokvm restart` on your NanoKVM to restart the service.

closed 2025-05-14 by krische