sipeed/NanoKVM #515: Unable to get NanoKVM to connect to network or get IP

jac2424, 2025-05-16:
I am having an issue with my NanoKVM with ATX board all of a sudden. It has been working fine for several weeks and now all of a sudden it does not seem to have a working network connection. Onboard screen shows no IP and my Unifi switch shows nothing detected on the port its plugged into. Tried different port on switch and cable and even tried to manually set port speed to 100 MBps full duplex but issue persisted. nanokvm was @2.2.7 and even tried reflashing image 1.4 but issue persisted. Any ideas how to fix this?

[image]

nippit, 2025-05-18:
I have exctly the same problem with two units 

issue 495
https://github.com/sipeed/NanoKVM/issues/495

not getting any response from sipeed or the comunity

nippit, 2025-05-18:
tried 
diffrent ports
diffrent switch
PoE switch
non PoE switch 
non managed switch

nope no connection!!!

jac2424, 2025-05-18:
I managed to get mine temporarily working by powering it on with no HDMI or other plugs connected, only power and Ethernet based on below thread. Seems like a common issue related to ground on Ethernet, regretting purchasing this thing as it just seems unpolished in hardware and software compared to competitors.

https://github.com/sipeed/NanoKVM/issues/26

nippit, 2025-05-18:
mine does NOT power on from power supply .

I use a battery pack to avoid all these possible ground issues 
nothing is connected power from battery pack and ethernet > no joy

I am an EEE (does understand the ground problems) YES I also regret I purchased two units if hardware has ground problems (IOT DOES ) it is  bad design and sadly you cannot fix it with software

Krazy998, 2025-05-20:
It may sound dumb - but i had a similar issue initially with no ip being set. I ended up using a different network cable (both cables were fine with different devices) However for some reason the other network cable seemed to work better. Its most likely related to the possible ground issues reported. However that worked for me.