sipeed/NanoKVM #517: CVE-2024-6387

pkolega, 2025-05-18:
Version of sshd (SSH-2.0-OpenSSH_9.6) is vulnerable. The vulnerability allows for remote code execution as root due to async-signal-unsafe functions being called in the SIGALRM handler.

Krazy998, 2025-05-20:
Consider using the debian image as a base. I have been using this for months now and as its effectively a debian image. All normal security and base image updates are easily covered:
scpcom has provided a debian image for the nanokvm which works perfectly. (I think it makes sense at some point to move to this to make it easier to maintain)

https://github.com/scpcom/sophgo-sg200x-debian

dormancygrace, 2026-09-14:
Hi! I've included a fix for this in [NanoKVM OS](https://github.com/dormancygrace/NanoKVM-OS), my independent community firmware for SG2002-based NanoKVM Cube and PCIe.

The beta-9 package manifest lists OpenSSH 10.5p1, beyond the 8.5p1–9.7p1 affected range documented in the [OpenSSH 9.8 security notes](https://www.openssh.org/txt/release-9.8). Fresh images also start with SSH disabled until enabled in settings.

This addresses the outdated OpenSSH component; it is not a claim of exploit reproduction on NanoKVM or of a complete security audit.

[Implementation / details](https://github.com/dormancygrace/NanoKVM-OS/blob/9f65f9554142db4a61ba866b4498363f4f21d711/firmware/release/licenses/manifest.csv#L50).

If you're still experiencing this issue, you're welcome to try it and let me know whether it resolves the problem on your setup.

It's currently in beta. Development testing has been on PCIe/UXC; Cube testing is still pending. Installing from stock firmware requires flashing a full SD image, so please back up your card first.

[Latest release](https://github.com/dormancygrace/NanoKVM-OS/releases/latest) · [Installation guide](https://github.com/dormancygrace/NanoKVM-OS/blob/main/docs/INSTALL.md)

Please include your board revision and reproduction steps when reporting results. Thanks!