sipeed/NanoKVM #523: 105-key keyboard support

TripleWhy, 2025-05-21:
When I press any key on my local keyboard, it appears to transmit the key, instead of the character. Which is fine, I just need to remember to type in the remote keyboard layout instead of the local one, or set them both to the same layout.

Now, if both the remote and local keyboards are 105-key layouts, you need that 105th key, as the characters on it aren't typically present on any other key. So here is my issue: NanoKVM does not transmit that 105th key. (Other keys work perfectly fine.)

alexislefebvre, 2025-07-20:
Is it the reason why the numpad on a full-size keyboard does not transmit data to the remote computer?

dormancygrace, 2026-09-14:
Hi! [NanoKVM OS](https://github.com/dormancygrace/NanoKVM-OS), my independent community firmware for SG2002-based NanoKVM Cube and PCIe, includes functionality relevant to this request.

The physical-key map includes the extra ISO keyboard key: browser code IntlBackslash is sent as USB HID usage 0x64. The keyboard path uses physical codes, so this key is not limited to a US 104-key layout.

The target OS keyboard layout must match the intended physical layout. This mapping does not by itself fix every international clipboard/paste or browser input issue.

[Implementation 1](https://github.com/dormancygrace/NanoKVM-OS/blob/9f65f9554142db4a61ba866b4498363f4f21d711/web/src/lib/keymap.ts#L137) · [Implementation 2](https://github.com/dormancygrace/NanoKVM-OS/blob/9f65f9554142db4a61ba866b4498363f4f21d711/web/src/lib/keyboard.ts#L9).

If this addresses your use case, you're welcome to try it and let me know how it works on your setup.

It's currently in beta. Development testing has been on PCIe/UXC; Cube testing is still pending. Installing from stock firmware requires flashing a full SD image, so please back up your card first.

[Latest release](https://github.com/dormancygrace/NanoKVM-OS/releases/latest) · [Installation guide](https://github.com/dormancygrace/NanoKVM-OS/blob/main/docs/INSTALL.md)

Please include your board revision and reproduction steps when reporting results. Thanks!