sipeed/NanoKVM #524: Host does not detect NanoKVM as video output.

NikIsAwesome, 2025-05-22:
Hi,

I have recently brought a PCIE NanoKVM, It was working fine when I tested it out the box, but after installing it in my server pc the KVM will not be detected as an display output in windows. I have tried multiple HDMI Cables, I have tried flashing different images to the SD card, I have tried different host pc. I have followed the guide on the FAQ and the issue still presists.

The HID and ATX power is working however, so its just an issue with the display.

brainkiller, 2025-05-27:
I have the same issue with the full NanoKVM on 2 different linux boxes.
I have the impression the video output to NanoKVM only shows up when a physical screen is attached and enabled at boot time.

Z2Z-GuGu, 2025-06-03:
> Hi,
>
> I have recently brought a PCIE NanoKVM, It was working fine when I tested it out the box, but after installing it in my server pc the KVM will not be detected as an display output in windows. I [...]

@NikIsAwesome Hello, can I understand that after powering on the NanoKVM-PCIe and directly connecting it to a host, the host does not recognize it as a display?

NikIsAwesome, 2025-06-09:
Hi, yes that is correct, the computer that I plug the HDMI into does not see the NanoKVM as a display output.

[image]

Z2Z-GuGu, 2025-06-13:
> Hi, yes that is correct, the computer that I plug the HDMI into does not see the NanoKVM as a display output.
> 
> [image]

In the web terminal, execute `cat /proc/cvitek/vi_dbg` to check if VIDevFPS is 0. This will quickly verify whether the NanoKVM is receiving a video signal from the HDMI interface. 

Additionally, if your host is connected to a real display, you can check the system's display settings to see if a second monitor is detected.

jbleuzen, 2025-06-14:
Hi,
I have the exact same problem. If I connect my computer to a screen and after I plug the nanoKVM PCIe, I have the video content on the webapp. If I boot directly with the nanoKVM plugged in I have no video output.

When directly booting with the nanoKVM plugged in, the `VIDevFPS` is 0.

NikIsAwesome, 2025-06-15:
> In the web terminal, execute `cat /proc/cvitek/vi_dbg` to check if VIDevFPS is 0. This will quickly verify whether the NanoKVM is receiving a video signal from the HDMI interface.
>
> Additionally, if [...]

After running the "cat /proc/cvitek/vi_dbg" command the VIDevFPS is 0. Also the host does recognise a second monitor as an output so it is not an issue with the host. Like I said in the first post I have tried multiple PC's and I have followed all the steps in the FAQ guide.

[image]

mangelajo, 2025-12-03:
I am experiencing the same issue, I tried 2.2.9 and 2.3.0

aletsch, 2026-04-05:
I'm having the same issue. I bought a brand new Pcie NanoKVM. I put it in my server, connected everything, but no matter what device i connect to (be my laptop or my server) i don't get any video signal at all.

Wen i i take a look inside the `vi_dbg` file as mentioned here it always displays a 0. No matter if i connect my server, my laptop or nothing. This sounds a lot to me that something is broken here. On my Laptop the kvm does not get recognised as a monitor at all.

``` bash
~ # cat /proc/cvitek/vi_dbg | grep VIDevFPS
VIDevFPS                :   0
```

When i test on my Laptop for screens connected when it is connected to the nanoKVM i do get nothing with `xrandr -q` (first one is the Laptop display)

```
Screen 0: minimum 320 x 200, current 1920 x 1080, maximum 16384 x 16384
eDP-1 connected primary 1920x1080+0+0 (normal left inverted right x axis y axis) 294mm x 165mm
   1920x1080     60.00*+  60.00    30.00  
   1680x1050     60.00  
   1400x1050     60.00  
   1600x900      60.00  
   1280x1024     60.00  
   1400x900      60.00  
   1280x960      60.00  
   1440x810      60.00  
   1368x768      60.00  
   1280x800      60.00  
   1280x720      60.00  
   1024x768      60.00  
   960x720       60.00  
   928x696       60.00  
   896x672       60.00  
   1024x576      60.00  
   960x600       60.00  
   960x540       60.00  
   800x600       60.00  
   840x525       60.00  
   864x486       60.00  
   700x525       60.00  
   800x450       60.00  
   640x512       60.00  
   700x450       60.00  
   640x480       60.00  
   720x405       60.00  
   684x384       60.00  
   640x360       60.00  
   512x384       60.00  
   512x288       60.00  
   480x270       60.00  
   400x300       60.00  
   432x243       60.00  
   320x240       60.00  
   360x202       60.00  
   320x180       60.00  
DP-1 disconnected (normal left inverted right x axis y axis)
DP-2 disconnected (normal left inverted right x axis y axis)
```

It looks like this when i connect a old monitor just as 
```
Screen 0: minimum 320 x 200, current 3600 x 1080, maximum 16384 x 16384
eDP-1 connected primary 1920x1080+0+0 (normal left inverted right x axis y axis) 294mm x 165mm
   1920x1080     60.00*+  60.00    30.00  
   1680x1050     60.00  
   1400x1050     60.00  
   1600x900      60.00  
   1280x1024     60.00  
   1400x900      60.00  
   1280x960      60.00  
   1440x810      60.00  
   1368x768      60.00  
   1280x800      60.00  
   1280x720      60.00  
   1024x768      60.00  
   960x720       60.00  
   928x696       60.00  
   896x672       60.00  
   1024x576      60.00  
   960x600       60.00  
   960x540       60.00  
   800x600       60.00  
   840x525       60.00  
   864x486       60.00  
   700x525       60.00  
   800x450       60.00  
   640x512       60.00  
   700x450       60.00  
   640x480       60.00  
   720x405       60.00  
   684x384       60.00  
   640x360       60.00  
   512x384       60.00  
   512x288       60.00  
   480x270       60.00  
   400x300       60.00  
   432x243       60.00  
   320x240       60.00  
   360x202       60.00  
   320x180       60.00  
DP-1 connected 1680x1050+1920+0 (normal left inverted right x axis y axis) 474mm x 296mm
   1680x1050     59.88*+
   1280x1024     75.02    60.02  
   1280x960      60.00  
   1152x864      75.00  
   1024x768      75.03    70.07    60.00  
   832x624       74.55  
   800x600       72.19    75.00    60.32    56.25  
   640x480       75.00    72.81    66.67    59.94  
   720x400       70.08  
DP-2 disconnected (normal left inverted right x axis y axis)
```

My System version is: v1.4.2
Application version is: 2.3.6

Did any one here got it working by now somehow?