sipeed/NanoKVM #526: [Feature Request]-solution is available!: Support German ISO layout keyboard (with AltGr modifier)

C2022H, 2025-05-23:
I have seen multiple issues from users using non-US keyboards (which have the ANSI layout).
Users with an ISO layout get wrong characters and some characters aren't reachable at all.
**I modified the code to support the AltGr modifier and allow a German ISO layout keyboard to work.**
**All keys work directly, with the virtual keyboard and with the paste function.**
Too bad, but I am not familiar with the "go" language and I cannot test other keyboards.
That is why my modified code is not 100% compatible with the ANSI layout or the other languages.
Especially the paste function requires changes in the go part of the code to stay compatible with the other keyboards.
I would really appreciate if somebody would take my code and include it into the main branch.
I didn't create a pull request, because my  code is not 100% compatible with the ANSI layout or the other languages.
You will find my code at
**https://github.com/C2022H/NanoKVM**
and the changes are in the **iso-de-keyboard** branch.

C2022H, 2025-05-29:
I am confused. For me it is working. Did anything change ? Any key behaving different as before? 
Which keyboard do you have?  ANSI or ISO layout? 
I have this one: https://de.sharkoon.com/product/SKILLER+MECH+SGK3
When I started, not only the AltGr and all keys that need this modifier did not work, but also the <> key (right to the left-shift key).
I set the Settings->Appearance to German. Then Keyboard to German as well. Do you see the German layout then for the on-screen keyboard?
I had issues that I got disk-full errors when I repeatedly uploaded NanoKVM-Server. Sometimes I had to power-off and power-on the device to get free space.

brkai, 2025-05-29:
Thanks for your reply!
I have an ISO layout (CORSAIR K55 RGB PRO) and i can see the german layout on the onscreen keyboard.
Just found out:
With ungoogled-chromium 137.0 the AltGR Key is recognized but the german umlaute don't work
(this is since the last flatpak update and i knew about the umlaute before).
With Firefox 139.0 AltGR is not recognized, everything else works fine.

I've tried different Browsers in the past (also LibreWolf for e.g.)  and they all behaved different:
Especially with LibreWolf i only was able to use an amercian layout.

I'll install a "normal" chromium and give you more feedback.

Is there an easy way to debug, what arrives at the server-component?

C2022H, 2025-05-29:
I never saw different behavior using different browsers.
I am on an openSUSE-15.6 Linux box and I tried 
Firefox 128.10.1esr
Chromium 136.0.7103.113 
Edge for Linux 136.0.3240.92
Regarding debugging: I am not familiar with the 'go' programming language at all, so I can't help you with this. Sorry.

brkai, 2025-05-29:
btw.:
THANKS FOR YOUR EFFORT!

ph0b0s101, 2025-06-02:
Thank you for working on this issue. I am here to vote for an integration in the main branch, so that it will integrated into the main project. German Kebboard would be soo nice.

C2022H, 2025-06-09:
> [web.modkai.zip](https://github.com/user-attachments/files/20521735/web.modkai.zip)
> 
> [@C2022H](https://github.com/C2022H) I played around with your sources a little bit and made a quick and dirty mod in index.tsx to make it work with firefox, edge and ungoogled chromium on my computer. the only browser where getModifierState("AltGraph") worked was chromium itself.
> 
> .... also added 'AltGraph' to mappings.ts
> 
> cheers kai

Hi kai,

I am still confused, that it isn't working for you. I've tried some more recent browsers (on Windows 11) and all browsers worked. As I did not get more negative feedback, I will leave my code like it is. But thanks for your patch, if more people have issues, this might help them out.