sipeed/NanoKVM #533: TIP: How to Wake a “Stubborn” PC with Wake-on-LAN and NanoKVM

supczinskib, 2025-05-27:
**TIP: How to Wake a “Stubborn” PC with Wake-on-LAN and NanoKVM**
_(Ubuntu + Intel I225-V NIC)_

Traditional Wake-on-LAN (WOL) only works when the network interface remains powered. On many motherboards with an Intel I225-V, issuing suspend puts the system into S3 and cuts power to the Ethernet port. Because Linux on this platform lacks S0ix support, ordinary WOL fails.

Below is a workaround that uses **NanoKVM**. The device listens for WOL “magic packets”; when it detects a packet addressed to the attached PC, it simulates pressing the **spacebar** on that machine’s virtual keyboard—effectively waking it up.

**Step-by-step**

- Switch the NanoKVM USB port to HID-Only mode.
- Note the MAC address of the network card in the PC you’ll be suspending.
- Edit /etc/inittab, adding one line: `wol::respawn:/root/wolfix.sh <MAC_ADDRESS> >/dev/null 2>&1`

- Create /root/wolfix.sh, make it executable (chmod +x), and paste the script unchanged:

```shell
#!/bin/sh

if [ $# -ne 1 ]; then
  echo "Usage: $0 <MAC>  # formats: aa:bb:cc:dd:ee:ff  aa-bb-cc-dd-ee-ff  aabbccddeeff" >&2
  exit 1
fi

MAC_RAW=$(printf '%s' "$1" | tr -d ':-' | tr '[:upper:]' '[:lower:]')

if [ ${#MAC_RAW} -ne 12 ] || echo "$MAC_RAW" | grep -q '[^0-9a-f]'; then
  echo "Invalid MAC: $1" >&2
  exit 1
fi

PREFIX=$'\xff\xff\xff\xff\xff\xff'
T=$PREFIX
i=0
while [ $i -lt 12 ]; do
  byte=${MAC_RAW:$i:2}
  T="$T"$(printf "\\x$byte")
  i=$((i+2))
done
export T

export R=$'\0\0\x2c\0\0\0\0\0\0\0\0\0\0\0\0\0'

exec socat -u UDP4-RECVFROM:9,reuseaddr,fork \
  SYSTEM:"dd bs=1 count=102 2>/dev/null | grep -Fq \"\$T\" && printf \"\$R\" > /dev/hidg0"
``` 

- **Reboot** the NanoKVM.

From now on, sending a WOL packet to the recorded MAC address makes NanoKVM “press” the spacebar, waking the computer even though its Ethernet port is unpowered. Ensure that **NanoKVM and the sleeping host are on the same subnet**.

> Who knows—maybe one day this feature will become native in the NanoKVM firmware. 🙂

hot22shot, 2025-06-26:
Thanks for this, I'll just adapt it to my use case : instead of pressing spacebar I want to start the PC (who's in shutdown state obviously).

susadmin, 2025-06-26:
Couldn't this be more integrated by issuing the 'Power' (on) command? Like on the header bar.