sipeed/NanoKVM #541: Feature request: custom h264 bitrate

corrpel, 2025-05-29:
It would be nice to have a custom bitrate option for h264.
Currently, the maximum in 5MB for lossless, and sometimes in looks pixelated for 1 second.

nwgat, 2025-07-12:
yeah, they need to improve the encoding, option to set to 25 will help alot

Filicya, 2025-07-14:
How to set a custom bitrate?

dormancygrace, 2026-09-14:
Hi! I've included support for this in [NanoKVM OS](https://github.com/dormancygrace/NanoKVM-OS), my independent community firmware for SG2002-based NanoKVM Cube and PCIe.

H.264/H.265 bitrate selection includes 15 and 20 Mbit/s CBR targets, in addition to the lower presets, in both Video settings and the toolbar.

These are selectable bitrate targets, not an arbitrary numeric bitrate editor or a guarantee of artifact-free playback.

[Implementation / details](https://github.com/dormancygrace/NanoKVM-OS/blob/9f65f9554142db4a61ba866b4498363f4f21d711/docs/RELEASE-beta-9.md).

If you're still experiencing this issue, you're welcome to try it and let me know whether it resolves the problem on your setup.

It's currently in beta. Development testing has been on PCIe/UXC; Cube testing is still pending. Installing from stock firmware requires flashing a full SD image, so please back up your card first.

[Latest release](https://github.com/dormancygrace/NanoKVM-OS/releases/latest) · [Installation guide](https://github.com/dormancygrace/NanoKVM-OS/blob/main/docs/INSTALL.md)

Please include your board revision and reproduction steps when reporting results. Thanks!