sipeed/NanoKVM #544: HID not working in bios (also not using HID-only mode)

Rainson12, 2025-06-02:
When entering BIOS / early boot it seems that HID is not working at all. When checking the /var/log/messages its spammed with:
```
Jun  2 11:47:46 kvm-d13c kern.info kernel: [  347.171628] dwc2 4340000.usb: wakeup: signalling skipped: is not allowed by host
Jun  2 11:47:46 kvm-d13c kern.err kernel: [  347.171646] configfs-gadget gadget: usb_gadget_wakeup
Jun  2 11:47:46 kvm-d13c kern.info kernel: [  347.189398] dwc2 4340000.usb: wakeup: signalling skipped: is not allowed by host
Jun  2 11:47:46 kvm-d13c kern.err kernel: [  347.189414] configfs-gadget gadget: usb_gadget_wakeup
Jun  2 11:47:46 kvm-d13c kern.info kernel: [  347.204958] dwc2 4340000.usb: wakeup: signalling skipped: is not allowed by host
Jun  2 11:47:46 kvm-d13c kern.err kernel: [  347.204974] configfs-gadget gadget: usb_gadget_wakeup
Jun  2 11:47:46 kvm-d13c kern.info kernel: [  347.221784] dwc2 4340000.usb: wakeup: signalling skipped: is not allowed by host
Jun  2 11:47:46 kvm-d13c kern.err kernel: [  347.221838] configfs-gadget gadget: usb_gadget_wakeup
Jun  2 11:47:46 kvm-d13c kern.info kernel: [  347.240372] dwc2 4340000.usb: wakeup: signalling skipped: is not allowed by host
Jun  2 11:47:46 kvm-d13c kern.err kernel: [  347.240387] configfs-gadget gadget: usb_gadget_wakeup
Jun  2 11:47:46 kvm-d13c kern.info kernel: [  347.255072] dwc2 4340000.usb: wakeup: signalling skipped: is not allowed by host
Jun  2 11:47:46 kvm-d13c kern.err kernel: [  347.255088] configfs-gadget gadget: usb_gadget_wakeup
Jun  2 11:47:46 kvm-d13c kern.info kernel: [  347.271637] dwc2 4340000.usb: wakeup: signalling skipped: is not allowed by host
Jun  2 11:47:46 kvm-d13c kern.err kernel: [  347.271653] configfs-gadget gadget: usb_gadget_wakeup
Jun  2 11:47:46 kvm-d13c kern.info kernel: [  347.288373] dwc2 4340000.usb: wakeup: signalling skipped: is not allowed by host
Jun  2 11:47:46 kvm-d13c kern.err kernel: [  347.288388] configfs-gadget gadget: usb_gadget_wakeup
Jun  2 11:47:46 kvm-d13c kern.info kernel: [  347.305364] dwc2 4340000.usb: wakeup: signalling skipped: is not allowed by host
Jun  2 11:47:46 kvm-d13c kern.err kernel: [  347.305383] configfs-gadget gadget: usb_gadget_wakeup
Jun  2 11:47:46 kvm-d13c kern.info kernel: [  347.323384] dwc2 4340000.usb: wakeup: signalling skipped: is not allowed by host
Jun  2 11:47:46 kvm-d13c kern.err kernel: [  347.323400] configfs-gadget gadget: usb_gadget_wakeup
```

This is how it looks in in gui:

[image]

Running on: v1.4.0 & 2.2.8 Web App

Rainson12, 2025-06-02:
i did several times, doesnt seem to work

svatoslove, 2025-06-02:
> i did several times, doesnt seem to work

Without mount ISO dont work too?

Z2Z-GuGu, 2025-06-13:
> I have the same issue. I see those same errors in the log.
> 
> Tried HID only mode, tried creating /boot/BIOS, tried different USB cables/ports, tried the power off > reboot kvm > power on order of events, and nothing works. HID light is not on and cat /sys/class/udc/4340000.usb/state returns "not attached".
> 
> Anything else I can try? The device is mostly useless to me if I can't do keyboard input.

If the HID status indicator is not lit, it indicates that the NanoKVM is not connected to the host's USB. This may be due to an unstable Type-C connection. Please try replacing the cable.

jlinton, 2025-07-16:
Same here, EDK2 based firmware, the HID keyboard is only recognized in Linux. I have to have another physical keyboard attached to navigate the BIOS/Grub2 menus. I might take a hack at this if it reproduces on the rpi4 at some later date.

jlinton, 2025-07-19:
Note a keyboard that works in the BIOS: 

```
Bus 009 Device 002: ID 046d:c31c Logitech, Inc. Keyboard K120
Negotiated speed: Low Speed (1Mbps)
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 [unknown]
  bDeviceSubClass         0 [unknown]
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x046d Logitech, Inc.
  idProduct          0xc31c Keyboard K120
  bcdDevice           64.00
  iManufacturer           1 Logitech
  iProduct                2 USB Keyboard
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x003b
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          3 U64.00_B0001
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower               90mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      **bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              2 USB Keyboard**
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.10
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 (null)
          wDescriptorLength      65
          Report Descriptors: 
            ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval              10
```

And the sispeed:

```
Bus 005 Device 002: ID 3346:1009 sipeed NanoKVM
Negotiated speed: High Speed (480Mbps)
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 [unknown]
  bDeviceSubClass         0 [unknown]
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x3346 sipeed
  idProduct          0x1009 NanoKVM
  bcdDevice            5.10
  iManufacturer           1 sipeed
  iProduct                2 NanoKVM
  iSerial                 3 0123456789ABCDEF
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0080
    bNumInterfaces          4
    bConfigurationValue     1
    iConfiguration          4 NanoKVM
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower              120mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 [unknown]
      bInterfaceProtocol      1 Keyboard
      iInterface              5 HID Interface
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.01
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 (null)
          wDescriptorLength      63
          Report Descriptors: 
            ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               4
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               4
```

And the AI response about this:

"The USB boot interface subclass refers to a specific usage of the USB interface class (01h) that allows devices to be used for booting a computer. This subclass, typically associated with Human Interface Devices (HID), enables keyboards and mice to be recognized by the computer's BIOS or UEFI firmware during the boot process. 
Here's a more detailed explanation: 

    USB Interface Classes:
    USB devices are categorized into classes based on their functionality. The interface class for Human Interface Devices (HID) is 03h.
    Subclass 01h (Boot Interface Class):
    Within the HID class, subclass 01h is specifically designated for boot interface devices, primarily keyboards and mice.
    Bootability:
    The BIOS or UEFI firmware of a computer can recognize and utilize devices belonging to the boot interface subclass during the startup sequence, allowing users to interact with the system before the operating system loads."

So, there is a simplified boot endpoint which is being used by BIOS and USB unaware OS's, that the sispeed isn't reporting.

mfmoros, 2025-08-04:
Hi, any idea when the hint proposed by @jlinton will be implemented?