sipeed/NanoKVM #553: Firmware 1,40 not working on NanoKvm Pcie

Muadib557, 2025-06-12:
Have NanoKvm Pcie on a Dell R710   with firmware 1.30 and it works reasonably well.  It doesn't show  the bios splash or boot information but it does allow me to remote boot the server which is what I wanted.
Other than the no bios screen, I've tried to update the firmware to 1.40 and it simply doesn't work.  I can't connect to the KVM, the connection is not accepted or simply not available.  I tried modifying the .cnf file with the IP of the KVM and it just gives a blank screen.  I  tried reflashing the TF card several times but to no avail.  The Kvm seems to boot fine with the small screen showing the correct IP address but it still doesn't work.

Any ideas would be helpful.

hasilhd, 2025-06-14:
Hi, i buyed PCIe verzion and came to me with preinstalled v1.40. Working, but also have problem to see boot screen, enter to BIOS. But, when OS (Windows, Debian) boot, all is OK. Also i can remotely turn on motherboard. Try clean reinstall.

marian790, 2025-06-14:
Hi, i buyed PCIe verzion and came to me with preinstalled v1.40. Working, but also have problem to see boot screen, enter to BIOS. But, when OS (Windows, Debian) boot, all is OK. Also i can remotely turn on motherboard. Try clean reinstall.

Muadib557, 2025-06-14:
I have tried re-flashing the TF card, re-downloaded the 1.4 firmware, reset the KVM, removed power and then re-inserted the PCIe.  Nothing seems to work.  I guess the R710 is just too old to behave as I want it to.  At least I can power it on and off which was my primary concern/
Just a note, I updated the software to the latest version (2.9)  It didn't help, but now the power button no longer lights up when I turn on the remote computer.

Muadib557, 2025-06-14:
I just re-flashed the TF card.  I still get " This site can't be reached #### refused to connect.

marian790, 2025-06-14:
Broken HW?

Muadib557, 2025-06-14:
Oh well.  I just re-inserted the original flash card with 1.30 and it works like before.  Now the power lite indicator is working again.  So I'm fine with it .
Thanks for all the suggestions.

RadegasTCZ, 2025-06-21:
I am trying to also understand what the issue is. But think I might be able to undestand, but to verify.

1. When you flash v1.3 firmware to the SD Card, the NanoKVM-PCIe is able to start up, but it cannot capture the whatever video comes out of the target computer firmware.
2. When you flash v1.4 firmware to the SD Card, the NanoKVM-PCIe just shows no signs of life?

My current set-up on a traditional desktop computer (ASUS Prime B450-PLUS, NVIDIA GT1030 as a video source), has flakey USB (ordered myself a galvanic USB isolator for HID to see what happens), but seems to work fine. If that helps you have a point of comparison.

[image]

Muadib557, 2025-07-18:
> Do you mean that the BIOS screen is not displaying but the system can show it after booting, regardless of whether the firmware version is 1.3.0 or 1.4.0? This might be due to the HDMI outputting a resolution that the NanoKVM does not support during the BIOS stage, which complicates the issue. I'm considering writing an EDID editor to improve this.
>
> Or are you referring to the inability to access the 1.4.0 firmware in the web interface, with even the login page not displaying correctly?
>
> That would be amazing @Z2Z-GuGu - I'm currently trying to rebuild the code with a new EDID as I'm having several problems which are related (1080p60 not working, and BIOS screens not working either). An EDID emulator fixes all the BIOS issues, but not 1080p60. I did managed to insert this EDID into the code and recompile: https://github.com/6by9/CSI2_device_config/blob/master/1080P50EDID.txt - that means the highest resolution/refresh rate is now 1080p50 if I understand the way EDID works correctly. That fixes my high resolution display issues, but still can't see the BIOS. I'm trying to patch the Established Timing I & II fields to fix this, but so far no success. I'm documenting my journey here as I hope it will help others: #571

The bios doesn't display on either 1.3 or 1.4.  The 1.4 firmware does is not reachable via the web interface.   I've tried all of the various resolutions etc. And no luck.  It's probably the old Dell server I'm using.  I'll keep trying with future updates, but as I mentioned. I'm just happy to be able to start and stop the server when I need to.   I'll keep checking for firmware updates.
Thanks

Sent from AT&T Yahoo Mail on Android

BlueBackbite, 2025-09-07:
Seeing similar issues with the NanoKvm cube, flashing 1.3 works, but when it updates to 1.4, no signs of life from it, even before connecting any display out. The device takes an extremely long time simply to respond to the root :80/ (~3500 ms) and then the rest doesn't load. When SSH is enabled, it times out before being able to do anything useful post login.