sipeed/NanoKVM #558: Why is it so useless?

CWKSR, 2025-06-20:
I dont get what im doing wrong. Ive implanted into the server connected power, hdmi, usb, network cable, static ip.

Im looking at a black window. But i have to exit the site and revisit to see the picture, why cant it auto refresh? It seems like it doenst notice if there is a picture... its absolutly useless this way?

What am i doing wrong?

razuritta, 2025-06-21:
It would be better to summarize the issue in the title.
like: Cannot view hdmi video input from the webpage, etc.

CWKSR, 2025-06-21:
Its not like i cant see it, the problem is i have to do something to see it, its not enough to be watching the webpage, what would have been my understanding of a KVM???!?!?!?!