sipeed/NanoKVM #562: Update failing Lib download returns 404

tudalex, 2025-06-23:
Hi I am trying to update from 2.1.4 to 2.2.8 and the updater always failing at trying to download the library with a 404 error.
I even tried the unofficial one and it does the same, but at least provides more details

Library download failed: 404 Client Error: Not Found for url: https://maixvision.sipeed.com/api/v1/nanokvm/encryption?uid=d1ca961f1359c262

Is this expected?

leehowley3, 2025-06-28:
Same issue here. I know I can update manually, but I would rather not.

leehowley3, 2025-06-28:
So what I ended up having to do was SSH into nanoKVM from another machine. From there I used curl to download the latest update from their cdn. 

`curl "https://cdn.sipeed.com/nanokvm/latest.zip?n=1751137490" --output latest.zip`

I then used unzip to unzip it.

`unzip latest.zip `

changed directories into latest/system

`cd latest/system`

Then ran the python script in there. 

`python update-nanokvm.py`

tudalex, 2025-06-29:
Thanks, this worked.

closed 2025-06-29 by tudalex