sipeed/NanoKVM #567: NGINX Reverse Proxy

megaman49, 2025-07-04:
I am trying to access my NanoKVM via NGINX Reverse Proxy Manager which is on another device on my network.  I can access the NanoKVM fine using the device's IP address, but I get a "server error (500)" any time I try to access it from an external network.  It does show the redirect to [mydomain]/account/login, and I don't get any errors on the NGINX device logs.

How do I download the error logs from the NanoVWM? Is there a guide for setting up the reverse proxy? 

I'm on the current 1.4 image updated to the KVM app 2.2.9

susadmin, 2025-07-05:
I use HAProxy which works (until the video streaming randomly stops but that is a different issue).

Probably best to setup nanokvm on plain http and then use your proxy to add ssl if that is what you want.

megaman49, 2025-07-05:
> I use HAProxy which works (until the video streaming randomly stops but that is a different issue).
> 
> Probably best to setup nanokvm on plain http and then use your proxy to add ssl if that is what you want.

I don't think that is the issue, I tried running my proxy as HTTP without SSL and I still get the same error.  I'm fairly certain the issue is a pathing problem.  When I enter the IP directly into the browser, the redirect goes to /#/auth/login, but when I enter the proxy domain, the URL redirects to /account/login and gives the error 500.

susadmin, 2025-07-05:
I just forward everything (that matches my URL) from my proxy to port 80.

I don't know if it helps but after I login (/#/auth/login), the base url is just https://<my url>/#/