sipeed/NanoKVM #575: Lock the keyboard in fullscreen mode

aarcro, 2025-07-09:
When going to full screen you should call `navigator.keyboard.lock();` so that the escape key and other special keys will function in the client.

hafeeshf, 2025-07-12:
how to do this ? permanently . for my nanokvm ?

ejbest, 2025-07-13:
I think am starting to understand what you were explaining the other day

aarcro, 2025-07-14:
> how to do this ? permanently . for my nanokvm ?

Right now I'm doing it manually in the JS console after reloading. The ask here is to have the client app call it when entering fullscreen. I might get to PR for that this week.

aarcro, 2025-07-15:
Here's another tip, Add this in a bookmark: `javascript:navigator.keyboard.lock();`

alexislefebvre, 2025-07-20:
I confirm that Alt+Tab is passed to the host and not the remote computer. Same thing for the windows key, which is useful on a Linux Live (graphical) system.

Since we can use the NanoKVM toolbar to toggle the fullscreen mode, there is no "risk" of being locked without exiting the NanoKVM interface.

wj-xiao, 2025-12-31:
Thank you all for your help. We've included this solution in [2.3.1](https://github.com/sipeed/NanoKVM/blob/main/CHANGELOG.md#231-2025-12-26).